RIOT-OS/RIOT #3606: RIOT-OS on emulated ARM device

Lexandro92, 2015-08-11:
I am very new to embbeded systems or RIOT, I'm an intern at a small IT company and my boss gave me the following task:
Create an emulated ARM environment (and kind of ARM board is OK) and create a CoAP, 6lowpan client/server on these emulated boards. So far I used QEMU and I have been trying to just run the example hello-world program on a Cortex-M3 environment, Cortex-A9 and riotboard the last 2 in the hopes that these are all ARMv7 and maybe they are competible. None of tese worked.
I've seen https://github.com/RIOT-OS/RIOT/issues/3377
does this mean that currently this is not possible? That we have to buy a board that is supported?

kaspar030, 2015-08-11:
AFAIK noone has done this, but I'd be really looking forward to it.

Cortex-A9 and riotboard are ARM*v*7, while our supported ARM7 MCUs are ARM*v*4, don't let ARMs naming confuse you.

OlegHahm, 2015-08-11:
Hi @Lexandro92 and welcome to RIOT! Yes, you're right there's currently no out of the box support for a virtual Cortex-M* platform. I have no clue how much work it would actually be to implement #3377. Maybe @gebart or @phiros have more insights. 

Regarding Cortex-A9 (ARMv7) platforms: there has been some initial work to port RIOT on boards like this (e.g. the RIoTboard (see https://github.com/RIOT-OS/RIOT/pull/1411), but more as a proof-on-concept or research purposes. In general platforms that can run a proper Linux are considered mostly out-of-scope for RIOT.

Lexandro92, 2015-08-11:
Thank you for all the quick anwsers!
So I've talked with my boss and this is what I understand why we need emulation so badly.
We want a project incredibly similar to this:
http://watr.li/about/
But with the addition that we don't only want this network to communicate with the server board, we want it to communicate with a cloud. And my boss really wants the emulation, becouse he doesn't want it to work for just one type of ARM board (like in the link Raspberry Pi and Samr21)  he wants it to work for all kinds of ARM boards...for reasons.
So is it possible to emulate any kind of ARM board that runs RIOT?
Does this make any sense? Sometimes he doesn't give a full explenation :D

Kijewski, 2015-08-11:
Well, RIOT does work for a wide range of ARM boards already: https://github.com/RIOT-OS/RIOT/wiki/Family%3A-ARM. I do not quite see how a QEMU emulation would be helpful to you, since your idea should work right out of the box. Did you look into @watr-li's source code?

gebart, 2015-08-11:
@Lexandro92 I do not understand why you need emulation as opposed to running the `native` target inside Linux. Simply getting RIOT to run inside QEMU will not make it possible to use external GPIOs or communication busses from inside the emulated environment.
My own use case for wanting the emulated QEMU Cortex M3 target is to be able to do continuous integration and test low level code in a controlled, automated manner.

*Do you plan on running these applications solely on emulated hardware?* or do you want the emulated machines only to use while developing to validate your design, (maybe the hardware is not even finished yet?) and then move the application to a physical board when the hardware has been manufactured?

For development of applications, algorithms and network communication you could start with the native platform, which runs RIOT as a Linux process (or OSX, or FreeBSD etc.), and then buy some evaluation boards which are similar to whatever your target hardware is (maybe you could find a board which uses the same CPU as your target board, for example). With RIOT, because of its hardware abstraction layer, it is quite easy to move application level code between different boards. Even the device drivers (external devices, not in-CPU modules) are abstracted to a point where it is possible to write a driver once and run it on many different boards.

closed 2019-09-10 by stale[bot]