RIOT-OS/RIOT #3636: cpu: k60: not using uart_stdio

kaspar030, 2015-08-14:
The k60 is using it's own newlib implementation.

Just making it use the shared newlib module would drop filesystem/fd support from the port, but the port-specific syscalls.c doesn't use uart_stdio, and also it still offers uart0 support.

Could we either

1. make that implementation use uart_stdio and drop uart0
2. factor out the syscalls into its own module, so the user can chose between the shared newlib implementation and the specific one
3. ?

@jfischer-phytec-iot ideas?

gebart, 2015-08-14:
I accidentally hid the cleanup commit inside https://github.com/RIOT-OS/RIOT/pull/3485. See https://github.com/gebart/RIOT/commit/ea019b308974f98a281247c5c798ee295df455eb
I'll open a separate pull for that commit.

gebart, 2015-08-14:
The whole FD stuff should be refactored and introduced properly on system level instead of being board specific. Anyone implementing this on RIOT can always use the k60 syscalls as a base for their implementation.

PR #3637 (2015-08-16): boards/mulle: cleanup
Remove old syscalls implementation from before cortexm_common unification and sys/newlib introduction.

fixes #3636 

Moved the power pin initialization before clock init to allow peripheral devices to perform power-on initialization while we are simply wasting time spinning in order for the clock to stabilize.
closed 2015-08-16 by gebart