RIOT-OS/RIOT #3638: RFC: Future of ng_networking

authmillenon, 2015-08-15:
Thinking about a stack independent connection API (see #3615) and POSIX sockets and the final restructuring of GNRC for the last few days got me thinking about the future of the `ng_networking` app. Basically we have two paths we can go with it:

1. Keep it a pure GNRC example to keep it simple. 
2. Rewrite it as a `conn` or even socket example, since `netapi` never was meant to be an end-user API. 

For the first case I would rename the app to gnrc or gnrc_networking, for the second I would remove the ng_ prefix.

PeterKietzmann, 2015-08-17:
I didn't look into #3615 so I can't rate the complexity. But to state my general opinion: I think it is mandatory to have an example that "presents" the whole stack, including the application layer API. Dependent on the parallels between to current "pure" `ng_networking` example and the new one, I would decide if it is reasonable to keep both applications in the master or not.

cgundogan, 2015-10-03:
What's the current state of this issue? Can it be closed?

authmillenon, 2015-10-03:
It'd like to have a little more discussion on that. Not just my opinion and @PeterKietzmann's ;)

kaspar030, 2015-10-03:
+1 for the first route, rename and create a new one using conn.

gebart, 2015-10-03:
I agree with @kaspar030, create a new example for conn, keep the gnrc_networking example for easier testing of the gnrc stack.

miri64, 2016-10-17:
Maybe I find some time until the release for this one.

miri64, 2016-10-31:
Mh somehow it seems that I missread the comments. The rename to gnrc_networking already happened so let's close it :-).

closed 2016-10-31 by miri64