RIOT-OS/RIOT #3649: CI: integrate `make scan-build` into build tests?

OlegHahm, 2015-08-18:
I wonder if it would make sense to integrate the `scan-build` target, introduced in #3468 and using clang's static code analyzer, into the build tests on the CI system. If yes, would just failing on an error report be enough?

phiros, 2015-08-23:
Failing on an error because the static code analyzer thinks there is something wrong in the code is IMHO not a good idea. 
I've seen quite a few false positives whilst using scan-build...

OlegHahm, 2015-08-23:
Well, we have already cpp-check in the build test which has a comparable risk of false positives. Do you have an example for a false positive with the clang-analyzer?

phiros, 2015-08-23:
Hmm don't remember exactly. Seems like when we got rid of the old network stack we also get rid of some of those issues. I would have to look into some older revs in order to find those issues again. 
However, the [FAQ](clang-analyzer) of clang-analyzer has some examples.

closed 2019-09-10 by stale[bot]