RIOT-OS/RIOT #3652: bit define inconsistent

katezilla, 2015-08-18:
As i was looking at a porting for a new board i stubled on inconsistent led-pin defines:

one way is to set the acutal bit, like for example in [ stm32f4discovery/include/board.h:55] (https://github.com/RIOT/boards/stm32f4discovery/include/board.h)
`#define LD3_PIN             (1 << 13)`

the other way i found just had the bit position decimal  like [nucleo-f334/include/board.h:57] (https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-f334/include/board.h)
`#define LED_GREEN_PIN       (5)`

This does not reduce functionality as the defines to set on, off, toggle the LEDs was implemented right.

Maybe for the future we should use a consistent way on defining a bit ?

PeterKietzmann, 2015-08-18:
I think the overall question, that does not only regard to the LED macros, is: Should we place a preferred coding style to the coding conventions or the development procedures, that describes how we should set one bit of a variable/register. One way would be:
```
x |= 0x4;
```
An other way would be:
```
x |= (1 << 2);
```
Or do "we" actually __not__ care? I think it is not reasonable to adapt all implementations to one style but it's fine with me to state a preferred way.

BytesGalore, 2015-08-18:
I'm in favour of: 
``` C
x |= (1 << 2);
```
but anyhow, +1 to state a preferred way to steer coming implementations to one direction

kaspar030, 2015-08-18:
I also prefer ```x |= (1 << 2);```, but i wouldn't mind *not* having this in the conventions.

OlegHahm, 2015-08-18:
So, you're not saying that you wouldn't mind not having this not in the conventions?

kaspar030, 2015-08-18:
Yep. Don't care. 

Please, if you decide to make this a convention, make clear that this does not affect any value just because it happens to be a power of two.

PeterKietzmann, 2015-08-19:
@katezilla, so what will you make out of this :-) ?

OlegHahm, 2015-09-03:
@katezilla, ping!

PeterKietzmann, 2015-09-03:
AFAIK @katezilla is on holiday for the next 2 weeks or so

OlegHahm, 2015-09-03:
thanks for the info

katezilla, 2015-09-24:
sry for the long delay. it would be nice to have a prefered way for the future, but not nessecary to put it in conventions and change it everywhere.
personaly i like `x |= (1 << 2);` the folowing lines using the define will be shorter :)

PeterKietzmann, 2015-09-25:
Now that we discussed our opinions I think we can close this one.

closed 2015-09-25 by PeterKietzmann