RIOT-OS/RIOT #3662: Tracker: RPL missing features (2015 edition)

gebart, 2015-08-19:
This is a tracker for any issues related to the new RPL implementation. See also #3417  

 - [ ] Length checks of packets in all receive functions (rpl_control_messages.c)
 - [ ] Contiki interoperability (Contiki as RPL tree root node, RIOT as router)
   - [ ] IPv6 Hop-by-hop RPL option, #3399 (workaround: reconfigure Contiki to not add HBH option)
   - [ ] Support global RPLInstanceIDs in DAO messages #3659 (no workaround)
   - [ ] Add ETX/MRHOF objective function support (workaround: reconfigure Contiki to use RPL_CONF_OF rpl_of0)
 - [ ] Non-storing mode (requires source route information)

Feel free to update and add points above.

cgundogan, 2015-08-19:
I added the DAG-Metric container option to the list, because I think that Contiki is using this. Am I right with this assumption @gebart ?

gebart, 2015-08-19:
@cgundogan From looking in the Contiki code, I think DAG Metric Container might be used for ETX. I don't have any more information though, sorry.

kYc0o, 2016-07-26:
Somehow this was completely forgotten... I hope it can be addressed for the next release...