RIOT-OS/RIOT #3664: cortexm_common: use of r13 is deprecated

authmillenon, 2015-08-19:
After the merge of #3333 I get the following warning:

```
$ BOARD=iot-lab_M3 IOTLAB_NODES=2 make -C examples/gnrc_networking/ all
make: Entering directory `/home/martine/Repositories/RIOT-OS/RIOT/examples/gnrc_networking'
Building application "gnrc_networking" for "iot-lab_M3" with MCU "stm32f1".

"make" -C /home/martine/Repositories/RIOT-OS/RIOT/boards/iot-lab_M3
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/core
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/stm32f1
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/cortexm_common
/tmp/ccwJhJJ3.s: Assembler messages:
/tmp/ccwJhJJ3.s:211: Warning: use of r13 is deprecated
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/cpu/stm32f1/periph
[…]
```

gebart, 2015-08-19:
I will look into this.

thomaseichinger, 2015-08-19:
The line causing this is [vectors_cortexm.c:137](https://github.com/RIOT-OS/RIOT/blob/master/cpu/cortexm_common/vectors_cortexm.c#L137).

`cmp r0, sp` doesn't seem suspicious on first sight.

OlegHahm, 2015-08-19:
I think `make scan-build` was reporting this since it was introduced.

thomaseichinger, 2015-08-19:
Oops, `r13` is `sp`. Forgot about this. :grimacing:

daniel-k, 2015-08-19:
I found [this](https://sourceware.org/ml/binutils/2009-02/msg00298.html) after googling. But I couldn't find the ARM docs yet, that mention this.

thomaseichinger, 2015-08-19:
Found this too, `cmp sp, r0` would be allowed.

gebart, 2015-08-19:
yes, the solution is to mov r1, sp first, then cmp r0, r1

thomaseichinger, 2015-08-19:
Some register juggling. :smile:

thomaseichinger, 2015-08-19:
As [reference](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0489g/Cihiddid.html)
> **Use of SP in ARM and Thumb instructions**
> You can use SP for Rn in ARM and Thumb instructions.
> You can use SP for Rm in ARM instructions but this is deprecated in ARMv6T2 and above.
> You can use SP for Rm in a 16-bit Thumb CMP Rn, Rm instruction but this is deprecated in ARMv6T2 and above. Other use of SP for Rm is not permitted in Thumb.

closed 2015-08-20 by PeterKietzmann

authmillenon, 2015-08-20:
@PeterKietzmann please explain why you closed this issue.

PeterKietzmann, 2015-08-20:
It was automatically closed after I merged #3666. Isn't it fixed for you?

authmillenon, 2015-08-20:
Ah, sorry. Somehow I landed in #3356 and this wasn't merged yet.

PR #3666 (2015-08-20): cortexm_common: don't compare with r13/sp in assembly, fix #3664
Should fix #3664.

Please test this, as I have no hardware at hand atm.
commit 86af1969 (2015-08-20): Merge pull request #3666 from daniel-k/cortexm_r13_deprecated

cortexm_common: don't compare with r13/sp in assembly, fix #3664
commit cb4aec67 (2015-08-20): cortexm_common: don't compare with r13/sp in assembly, fix #3664