RIOT-OS/RIOT #3668: ICMPV6: Wrong Checksum for Packets with Payload < 10

cgundogan, 2015-08-19:
When I have an ICMPV6 packet with an (ipv6) payload of 8, then the icmp code complains about a wrong checksum ([gnrc_icmpv6.c](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/icmpv6/gnrc_icmpv6.c#L74)).
I can also notice a 2 bytes padding in wireshark at the end of the frame (and marked as `VSS-Monitoring ethernet trailer, Source Port: 0`)

Rationale: `DAO-ACK`s are only 8 byte of data if the `DODAG-ID` is omitted, but they never reach the `RPL` thread, because the icmpv6 code discards them.

cgundogan, 2015-08-19:
tested on native and with `#3659` by using a global instance id (`1 <= id <= 127`)

authmillenon, 2015-08-19:
That would explain some behavior I had yesterday when testing IPv6 raw sockets (but did not come to analyse it further). Can you maybe investigate this further? I don't know if I have the time for this this week :-/.

authmillenon, 2015-08-20:
Does #3669 fix it?

cgundogan, 2015-08-20:
Yes, with #3669 the checksums are calculated correctly and the `DAO-ACK`s get dispatched to thr `RPL` thread

closed 2015-08-20 by cgundogan