RIOT-OS/RIOT #3672: Ping6 on the iotlab-m3 times out for payload sizes >= 2000

cgundogan, 2015-08-20:
Pinging on the testbed (iotlab-m3) w/ and w/o iphc times out for payload sizes `>= 2000`.

`ping6 abcd::x 2000`  => timeout
`ping6 abcd::x 1999`  => works perfectly

cgundogan, 2015-08-20:
see discussion in #3671

OlegHahm, 2015-08-20:
Have you tried with size above 2000?

cgundogan, 2015-08-20:
Yes, ? `2001` and `3000` resulted in the same misbehaviour.

OlegHahm, 2015-08-20:
Btw. I cannot even ping google.com from my PC with payloads above 1500 bytes.

authmillenon, 2015-08-20:
@OlegHahm Look at the MTU of your Ethernet interface ;-)

authmillenon, 2015-08-20:
but speaking of: I think IPv6 isn't currently testing if the packet exceeds the interface's MTU :/

OlegHahm, 2015-08-20:
Link local works, so I guess its rather a question of the routers in between.

authmillenon, 2015-08-20:
Maybe they don't allow IPv6 fragmentation ;-)

authmillenon, 2015-08-20:
> but speaking of: I think IPv6 isn't currently testing if the packet exceeds the interface's MTU :/

We should fix this issue first however, to not let it reappear in the future.

cgundogan, 2015-08-20:
Debugging enabled in sixlowpan and pinging with a payload size `>=2000`: 
`dispatch e8 ... is not supported`. @authmillenon does it ring a bell to you?

authmillenon, 2015-08-20:
For some reason 6LoWPAN is receiving NALP frames.

authmillenon, 2015-08-20:
(maybe even sending them, but why oO)

authmillenon, 2015-08-20:
Are you receiving any other fragments beforehand?

authmillenon, 2015-08-20:
Looked it up right now: `0xe8` is a subsequent fragment dispatch, so … I need to investigate further, why this is not recognized.

OlegHahm, 2015-08-20:
@authmillenon,
> For some reason 6LoWPAN is receiving NALP frames.

That reminds me that the 6LoWPAN unittests revealed that the `is_nalp` function may give wrong results (or the test was broken). Let me check.

authmillenon, 2015-08-20:
m( I know what is happening.

authmillenon, 2015-08-20:
I was so focussed on the 2000, I forgot about the IPv6 and ICPMv6 header: With them the total datagram size is 2048, which is 2^11. The datagram size field of 6LoWPAN only fits 2047 at max [1].

[1] https://tools.ietf.org/html/rfc4944#section-5.3

authmillenon, 2015-08-20:
So 6LoWPAN just does not support `ping6 2000` ;)

authmillenon, 2015-08-20:
Didn't we had an issue like that a few weeks back?

authmillenon, 2015-08-20:
It's a bug however, that the frame is send out regardless.

PR #3674 (2015-08-21): gnrc_sixlowpan: don't send packets that exceed maximum datagram size
The datagram size field 6LoWPAN fragmentation only can handle values up to 2047. This PR ensures that a sender does not exceed this maximum packet length.

Fixes #3672.
closed 2015-08-21 by cgundogan