RIOT-OS/RIOT #3692: STM32L1 get_cpuid() hard faults when using a Cat. 1 or Cat. 2 STM32L1

DipSwitch, 2015-08-22:
From the STM32L1 Reference Manual (31.2 Unique device ID registers (96 bits)):

    Base address: 0x1FF80050 for Cat.1 and Cat.2 devices and 0x1FF800D0 for Cat.3, Cat.4, Cat.5 and Cat.6 devices

Three solutions possible for this problem:
* Compile time: Via the linkerscript for the device (this I would prefer since this is the cleanest solution in my opinion)

		MEMORY
		{
			rom (rx)        : ORIGIN = 0x08000000, LENGTH = 128K
			ram (rw)        : ORIGIN = 0x20000000, LENGTH = 32K
			
			cpuid (r)		: ORIGIN = 0x1FF80050, LENGTH = 12
		}

		_cpuid_address = ORIGIN(cpuid);

		INCLUDE cortexm_base.ld

* Runtime: read the DBGMCU_IDCODE register (address: 0xe004 2000) and use something like:

		#define STM32L1_DEV_ID_CAT1                 0x416
		#define STM32L1_DEV_ID_CAT2                 0x429
		#define STM32L1_DEV_ID_CAT3                 0x427
		#define STM32L1_DEV_ID_CAT4                 0x436
		#define STM32L1_DEV_ID_CAT56                0x437

		#define STM32L1_CPUID_ADDR_CAT12            (0x1ff80050)
		#define STM32L1_CPUID_ADDR_CAT3456          (0x1ff800d0)

		void cpuid_get(void *id)
		{
		#define ST_DEV_ID                           ((DBGMCU->IDCODE) & DBGMCU_IDCODE_DEV_ID)
			uint32_t cpuid_address = (ST_DEV_ID == STM32L1_DEV_ID_CAT1 || ST_DEV_ID == STM32L1_DEV_ID_CAT2)
										? STM32L1_CPUID_ADDR_CAT12
										: STM32L1_CPUID_ADDR_CAT3456;

			memcpy(id, (void *)(cpuid_address), CPUID_ID_LEN);
		#undef ST_DEV_ID
		}

* Compile time: By setting an define an extra define in `Makefile.include` or in `board.h`
    Here I would prefer an extra define in `Makefile.include` since you already place CPU information here.

		export CPU = stm32l1
		export CPU_MODEL = stm32l151rba
		export CPUID_LOCATION = 0x1ff80050

DipSwitch, 2015-08-25:
Fixed with release d2cad2a

cgundogan, 2015-08-25:
according to @DipSwitch, this has been fixed by #3701. So I close this.

closed 2015-08-25 by cgundogan