RIOT-OS/RIOT #3695: ipv6: Modules.dep not complete?

A-Paul, 2015-08-23:
In https://github.com/RIOT-OS/RIOT/pull/3349 I experienced that in `Makefile.include`

    USEMODULE += ng_ipv6

was not enough as I got compile error. Using the errors as a hint I added

    USEMODULE += ng_icmpv6
    USEMODULE += ng_ndp

which stopped compiler errors. Although this worked as a workaround maybe these additions would be better placed in `$RIOTBASE/Makefile.dep`

After trying to update with `s/ng_/gnrc_/` (I haven't pushed those changes now) I get new errors for undefined `gnrc_ndp_internal_sl2a_opt_handle`.

Before beginning to search for missing USEMODULE entries by trial & error again, maybe someone can investigate the situation. At least I'm not shure if the workaround was the correct approach.

PeterKietzmann, 2015-08-23:
@A-Paul, please add appropriate labels to an issue and assign a maintainer, if you have one in mind :-)

authmillenon, 2015-08-23:
1) its 'gnrc_ipv6' now
2) I can't really replicate this error. When I make 

```bash
USEMODULE=gnrc_ipv6 make -C tests/unittests/ tests-core
```

I don't get any errors.

PeterKietzmann, 2015-08-23:
Without looking into details I'd propose to rebase and be aware that #3661 is not merged now. Otherwise, is it a compiler dependant thing (I don't think so)? Which compilers are you using?

authmillenon, 2015-08-23:
Did it on current master, for native I use `gcc (Ubuntu 4.8.4-2ubuntu1~14.04) 4.8.4` and `Ubuntu clang version 3.4-1ubuntu3 (tags/RELEASE_34/final)`, on arm platforms I use `arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 224288]`

PeterKietzmann, 2015-08-23:
(Was more looking to Paul as `ng_ipv6` seems outdated ...)

A-Paul, 2015-08-23:
@authmillenon Although I'm one day late: Happy Birthday :cake:

> 1) its 'gnrc_ipv6' now

I know but as I wrote:

> After trying to update with s/ng_/gnrc_/ (I haven't pushed those changes now) I get new errors for undefined gnrc_ndp_internal_sl2a_opt_handle.

Shall I push this despite it's not compiling?

> 2) I can't really replicate this error. When I make

If you check out the PR and remove the additional USEMODULEs you'll see. I added them for a reason.

A-Paul, 2015-08-23:
@PeterKietzmann : gcc (Debian 4.9.2-10) 4.9.2

A-Paul, 2015-08-23:
Forgot to mention it. I had rebased before `s/ng_/gnrc_/`.

authmillenon, 2015-08-23:
Can you please rebase the PR first?

authmillenon, 2015-08-23:
It should compile without `gnrc_icmpv6` and `gnrc_ndp` then.

A-Paul, 2015-08-23:
@authmillenon See https://github.com/RIOT-OS/RIOT/issues/3695#issuecomment-133943572.

authmillenon, 2015-08-23:
What about it? I'm confused. Can you just rebase the PR so I can check out what is going on?

A-Paul, 2015-08-23:
Sorry, I confused something...

A-Paul, 2015-08-23:
Pushed.

authmillenon, 2015-08-23:
Ah saw it now myself. Don't know why it is not showing up in master, but https://github.com/RIOT-OS/RIOT/pull/3698 should fix it in a better way.

A-Paul, 2015-08-24:
Fixed by #3698.

closed 2015-08-24 by A-Paul