RIOT-OS/RIOT #3707: gnrc_ipv6.c can't send IPv6 packets with more than 1 users

DipSwitch, 2015-08-25:
When an IPv6 pktsnpt is used more than 1 time when it enters gnrc_ipv6.c:515 `gnrc_pktbuf_start_write` will create a new snip to fill fill the source address, and the checksum.

        /* seize payload as temporary variable */
        payload = gnrc_pktbuf_start_write(ipv6);

However if this happens `payload` is never set to the `pkt` variable which is used to send to the packet on line 592 `_send_unicast(iface, l2addr, l2addr_len, pkt);`

This line should be: `_send_unicast(iface, l2addr, l2addr_len, ipv6);` If i'm correct

DipSwitch, 2015-08-25:
This however will break the functionality for packets chains that start with type `GNRC_NETTYPE_NETIF` 
But maybe this doesn't mind because the packet is removed anyway in `_send_unicast`

DipSwitch, 2015-08-25:
This can be reproduced / tested with `examples/udp.c` and change line 83 from:

        gnrc_pktbuf_hold(ip, gnrc_netreg_num(GNRC_NETTYPE_UDP,
                                             GNRC_NETREG_DEMUX_CTX_ALL));

to:

        gnrc_pktbuf_hold(ip, gnrc_netreg_num(GNRC_NETTYPE_UDP,
                                             GNRC_NETREG_DEMUX_CTX_ALL) - 1);

DipSwitch, 2015-08-25:
Same goes for `gnrc_udp.c` where on line 160 `gnrc_netapi_dispatch_send(pkt->type, GNRC_NETREG_DEMUX_CTX_ALL, pkt)` is called with packet while packet is the original and not the replaced one by line 149 `udp_snip = gnrc_pktbuf_start_write(udp_snip);`

So line 160 should be: `if (!gnrc_netapi_dispatch_send(udp_snip->type, GNRC_NETREG_DEMUX_CTX_ALL, udp_snip)) {`

authmillenon, 2015-08-25:
That the packet is duplicated is intended, to have full write access to it.

DipSwitch, 2015-08-25:
I know that, but if you update the packet with write access, but both gnrc_ipv6.c and gnrc_udp.c send the old packet which isn't changed.

authmillenon, 2015-08-25:
Okay, can you update the issue's title accordingly?

DipSwitch, 2015-08-25:
So basically:

gnrc_udp.c:160 should be: `if (!gnrc_netapi_dispatch_send(udp_snip->type, GNRC_NETREG_DEMUX_CTX_ALL, udp_snip)) {`
gnrc_ipv6.c:592 should be: `_send_unicast(iface, l2addr, l2addr_len, ipv6);`

However with this change my UDP thread hanges if I send a packet with more users than there actually are.

authmillenon, 2015-08-25:
No. For UDP the IPv6 header would be missing and for IPv6 the netif header would be missing. The problem is a little bit more complex.

authmillenon, 2015-08-25:
Basically you have to duplicate every header up until the UDP/IPv6 header and update the packet tree accordingly.

authmillenon, 2015-08-25:
@haukepetersen do you have any idea how to do this with least effort?

authmillenon, 2015-08-25:
#3708 should at least fix IPv6. Will look into UDP as soon as I am at Hack'n'ACK.

DipSwitch, 2015-08-25:
Yeah I noticed, then there is another problem.

If you change the `examples\gnrc_networking\udp.c` to remove the - 1 the from the call to `gnrc_pktbuf_hold` on line 83 the UDP thread would hang itself. Since the message/packet is send to the UDP thread which would dispatch it to itself which receives it in the send function, pushes it to dispatch. etc. etc. etc.

authmillenon, 2015-08-25:
Well you are producing a packet that is never properly released this way, so this isn't that surprising surprising.

authmillenon, 2015-08-25:
Which TCP implementation are you fixing btw? As far as I know there is a student (@brummer-simon) currently working on his bachelor thesis for this.

DipSwitch, 2015-08-25:
My master is a merge of his branch and the RIOT-OS master I ported it to
gncr and fixed the outstanding issue

authmillenon, 2015-08-25:
Does the current version of #3708 fixes it for you? About the other stuff: I just wanted to let you know, that there is currently someone working on TCP and prevent work from being doubled ;).

DipSwitch, 2015-08-25:
I will check when I'm home. :)

Is he still working on it now? His branch was a bit outdated. I can help,
but I don't know if thats oke, since it's for his bachelor...

OlegHahm, 2015-08-25:
@DipSwitch, yes, he's still working on this and I guess help is appreciated.

authmillenon, 2015-08-25:
He is to my knowledge. Please communicate about this with him or one of his advisors (all of them should be reachable via the devel ML). As far as I know is TCP only a fall-out of his actual thesis topic, so there might be some help possible. :)

OlegHahm, 2015-08-25:
Just to be sure, I contacted his supervisor.

brummer-simon, 2015-08-25:
Okay short update from my side to clear things up. Yes my Branch is a bit outdated. Currently I am trying to get my local branch fixed(All the renaming during the last weeks) to get my current state to run on the gnrc stack. 

After that I will focus on my Bachelor Thesis and pause further developent. I will continue working on it, after finishing my Thesis. Everyone who wants to fix the current state or bring the development further, is more than welcome :). While I am working on my Thesis I am of course available for questions.

DipSwitch, 2015-08-25:
@brummer-simon If you merge my branch almost all renaming is done ;) branch `tcp_implement`

DipSwitch, 2015-08-25:
@authmillenon Now now UDP or TCP is comming out of the stack, ICMP works however

OlegHahm, 2015-08-25:
now now?

DipSwitch, 2015-08-25:
Now no =)

authmillenon, 2015-08-25:
With my tests UDP communication was possible. Can you describe me your test setup?

DipSwitch, 2015-08-25:
Sorry my mistake, linux with a tap-bridge doesn't properly come back from an hybernate. UDP works, now testing the rest

authmillenon, 2015-08-25:
Now only a maintainer needs to ACK the PR, since UDP and ICMPv6 are the only things I can currently fix in master (and IPv6 of course, but this is used by both)

DipSwitch, 2015-08-25:
This would also work for the TCP implementation I guess, packets seem good :+1:

authmillenon, 2015-08-26:
@OlegHahm @haukepetersen @kaspar030 @PeterKietzmann may one of you review this?

DipSwitch, 2015-08-30:
I was wondering, why would you want to write protect the headers? Can't we make a flag in the nettype or an extra flags byte to know if the headers are generated and checksums calculated?
Because for retry purposes you would eventually want to resend that packet before freeing it from the packet buffer. Once the headers are generated, they will never change. So if you 'write lock' the data after filling the headers you can just keep a queue of not ACK'ed packet buffers. (Not only for TCP but also TFTP for example, think firmware updates).
If you would force this convention system wide, you would never have to copy the buffers to temp buffers just to fill the header every packet user want to have eventually.

The only problem I can think of that maybe occur is if you want to send the packet out of multiple interfaces... multicast / broadcast. But maybe it would be enough to make an exception for those instead for handling all packets the same.

Just a brain fart, maybe I'm overlooking something...

closed 2015-08-31 by authmillenon

OlegHahm, 2015-08-31:
@authmillenon, do you have any comments on @DipSwitch's thoughts here?

authmillenon, 2015-08-31:
I had already posted a reply this afternoon, but somehow it disappeared Oo. Will reformulate an answer tomorrow.

PR #3708 (2015-08-31): gnrc(_udp|_ipv6): write protect preceding headers and hand actually hand them over on send
Fixes #3707 by write-protecting all headers up until the relevant header. For `gnrc_ipv6` this was already the case, `pkt` just needed to be reset. For `gnrc_udp` I copied all headers up until `udp_snip` while searching for `udp_snip` and updated the packet snip chain.
commit 8a07f8c5 (2015-08-31): gnrc_ipv6: fix #3707
commit f81f43ea (2015-08-31): gnrc_udp: fix #3707