RIOT-OS/RIOT #3727: gnrc_networking example: doesn't compile on OS X

emmanuelsearch, 2015-08-27:
the error thrown on OS X (tested on 10.10.5) is the following:

```
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Building application "gnrc_networking" for "native" with MCU "native".

/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/examples/gnrc_networking/udp.c:89:48: error: 
      format specifies type 'int' but the argument has type 'size_t'
      (aka 'unsigned long') [-Werror,-Wformat]
    printf("Success: send %i byte to %s:%u\n", payload->size, addr_str, tmp);
                          ~~                   ^~~~~~~~~~~~~
                          %zu
1 error generated.
make[1]: *** [/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/examples/gnrc_networking/bin/native/gnrc_networking/udp.o] Error 1
make: *** [all] Error 2
```

If one casts the size as an int or unsigned to correct the above, we get another slew of errors, shown below:

```
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:329:5: error: 
      'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Werror,-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:507:9: error: 
      'getcontext' is deprecated: first deprecated in OS X 10.6
      [-Werror,-Wdeprecated-declarations]
    if (getcontext(&native_isr_context) == -1) {
        ^
/usr/include/ucontext.h:37:6: note: 'getcontext' has been explicitly marked
      deprecated here
int  getcontext(ucontext_t *) __OSX_AVAILABLE_BUT_DEPRECATED(__MAC_10_5...
     ^
/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native/irq_cpu.c:525:5: error: 
      'makecontext' is deprecated: first deprecated in OS X 10.6
      [-Werror,-Wdeprecated-declarations]
    makecontext(&native_isr_context, native_irq_handler, 0);
    ^
/usr/include/ucontext.h:38:6: note: 'makecontext' has been explicitly marked
      deprecated here
void makecontext(ucontext_t *, void (*)(), int, ...) __OSX_AVAILABLE_BUT...
     ^
3 errors generated.
make[2]: *** [/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/examples/gnrc_networking/bin/native/cpu/irq_cpu.o] Error 1
make[1]: *** [ALL--/Users/emmanuelB/Documents/work/CODING/RIOT/RIOT/cpu/native] Error 2
make: *** [all] Error 2
```

thomaseichinger, 2015-08-27:
@emmanuelsearch the second cited errors are solved by https://github.com/RIOT-OS/RIOT/pull/3393.
clang seems to be picky on formating, question is, if we should turn warnings, and thous errors off, for these?

kaspar030, 2015-09-04:
I just tried on 

```Darwin milhouse.imp.fu-berlin.de 14.4.0 Darwin Kernel Version 14.4.0: Thu May 28 11:35:04 PDT 2015; root:xnu-2782.30.5~1/RELEASE_X86_64 x86_64```

Compiles fine. emmanuel, could you try again?

emmanuelsearch, 2015-09-05:
compiles fine on my machine too since #3393 got merged actually. So I'll close this.

closed 2015-09-05 by emmanuelsearch