RIOT-OS/RIOT #3735: tests/periph_gpio: not working correclty?

PeterKietzmann, 2015-08-28:
It seems that tests/periph_gpio does not work currently. There are no shell functions available. I tried it on samr21-xpro. Taday I won't find time to look into that.

```
2015-08-28 09:02:10,417 - INFO # GPIO peripheral driver test
2015-08-28 09:02:10,418 - INFO # 
2015-08-28 09:02:10,424 - INFO # In this test, pins are specified by integer port and pin numbers.
2015-08-28 09:02:10,429 - INFO # So if your platform has a pin PA01, it will be port=0 and pin=1,
2015-08-28 09:02:10,432 - INFO # PC14 would be port=2 and pin=14 etc.
2015-08-28 09:02:10,432 - INFO # 
2015-08-28 09:02:10,438 - INFO # NOTE: make sure the values you use exist on your platform! The
2015-08-28 09:02:10,443 - INFO #       behavior for not existing ports/pins is not defined!
2015-08-28 09:02:13,699 - INFO # > kernel_init(): This is RIOT! (Version: 2014.12-2959-gc144-kietzmann-Latitude-E6410-ibriano-samd21)
2015-08-28 09:02:13,703 - INFO # kernel_init(): jumping into first task...
2015-08-28 09:02:13,705 - INFO # GPIO peripheral driver test
2015-08-28 09:02:13,705 - INFO # 
2015-08-28 09:02:13,711 - INFO # In this test, pins are specified by integer port and pin numbers.
2015-08-28 09:02:13,717 - INFO # So if your platform has a pin PA01, it will be port=0 and pin=1,
2015-08-28 09:02:13,720 - INFO # PC14 would be port=2 and pin=14 etc.
2015-08-28 09:02:13,720 - INFO # 
2015-08-28 09:02:13,725 - INFO # NOTE: make sure the values you use exist on your platform! The
2015-08-28 09:02:13,730 - INFO #       behavior for not existing ports/pins is not defined!
help
2015-08-28 09:02:15,303 - INFO # > help
2015-08-28 09:02:15,306 - INFO # Command              Description
2015-08-28 09:02:15,309 - INFO # ---------------------------------------

```

kaspar030, 2015-08-28:
> There are no shell functions available.

I can confirm this, will take a look.

kaspar030, 2015-08-28:
Got it, could you test?

PeterKietzmann, 2015-08-28:
Fixed by #3736

closed 2015-08-28 by PeterKietzmann