RIOT-OS/RIOT #3770: avr8: assert compile error

kaspar030, 2015-09-03:
I've stumpled over this error a couple of times (in travis builds):

/home/travis/build/RIOT-OS/RIOT/sys/xtimer/xtimer.c: In function '_xtimer_sleep':

```
/home/travis/build/RIOT-OS/RIOT/sys/xtimer/xtimer.c:39:9: error: implicit declaration of function 'abort' [-Werror=implicit-function-declaration]

         assert(!long_offset);

         ^
```

It doesn't show up locally, and it seems to not always pop up in travis. Ideas?

cgundogan, 2015-09-03:
I had a brief offline discussion with @OlegHahm once .. It could be that travis uses a different toolchain, which does not define the abort function

OlegHahm, 2015-09-03:
Moving to #3722 could be a solution. ;)

authmillenon, 2016-02-09:
Is this issue solved, since RIOT has it's own assert now?

OlegHahm, 2016-02-09:
Haven't seen that recently. Feel free to re-open if you experienced otherwise.

closed 2016-02-09 by OlegHahm