RIOT-OS/RIOT #3772: cpu: kinetis_common: periph/timer using inefficient initialization

kaspar030, 2015-09-03:
While analyzing the impact of using xtimer all over the place, kinetis_common based boards use a lot more code:

```
text	data	bss	dec	BOARD/BINDIRBASE

-792	-4	-84	-880	airfy-beacon
11772	148	2748	14668	oldbin
10980	144	2664	13788	bin

-784	-4	-124	-912	arduino-due
12028	152	2824	15004	oldbin
11244	148	2700	14092	bin

-3328	-98	-46	-3472	arduino-mega2560
14740	818	808	16366	oldbin
11412	720	762	12894	bin

-780	-4	-68	-852	cc2538dk
11576	148	2740	14464	oldbin
10796	144	2672	13612	bin

-716	-4	-60	-780	ek-lm4f120xl
11492	148	2788	14428	oldbin
10776	144	2728	13648	bin

-768	-4	-100	-872	f4vi1
13044	152	2760	15956	oldbin
12276	148	2660	15084	bin

-728	-4	-92	-824	fox
17784	148	2924	20856	oldbin
17056	144	2832	20032	bin

1352	-4	-60	1288	frdm-k64f
15928	1244	4900	22072	oldbin
17280	1240	4840	23360	bin

-776	-4	-92	-872	iotlab-m3
18452	148	2932	21532	oldbin
17676	144	2840	20660	bin

-732	-4	-92	-828	limifrog-v1
12960	148	2764	15872	oldbin
12228	144	2672	15044	bin

-728	-4	-92	-824	mbed_lpc1768
11312	148	2764	14224	oldbin
10584	144	2672	13400	bin
-768	-4	-100	-872	msbiot
13324	152	2784	16260	oldbin
12556	148	2684	15388	bin

1376	-4	-60	1312	mulle
17808	1268	4628	23704	oldbin
19184	1264	4568	25016	bin

2252	4	-64	2192	native
55100	536	60672	116308	oldbin
57352	540	60608	118500	bin

-792	-4	-84	-880	nrf51dongle
11796	148	2748	14692	oldbin
11004	144	2664	13812	bin

-792	-4	-84	-880	nrf6310
11772	148	2748	14668	oldbin
10980	144	2664	13788	bin

-780	-4	-100	-884	nucleo-f091
18568	152	2776	21496	oldbin
17788	148	2676	20612	bin

-768	-4	-100	-872	nucleo-f303
13000	148	2780	15928	oldbin
12232	144	2680	15056	bin

-768	-4	-100	-872	nucleo-f334
12744	148	2756	15648	oldbin
11976	144	2656	14776	bin

-732	-4	-92	-828	nucleo-l1
12960	148	2756	15864	oldbin
12228	144	2664	15036	bin

-780	-4	-68	-852	openmote
11484	148	2740	14372	oldbin
10704	144	2672	13520	bin

1352	-4	-60	1288	pba-d-01-kw2x
15996	1244	4284	21524	oldbin
17348	1240	4224	22812	bin

-792	-4	-84	-880	pca10000
11796	148	2748	14692	oldbin
11004	144	2664	13812	bin

-792	-4	-84	-880	pca10005
11776	148	2748	14672	oldbin
10984	144	2664	13792	bin

-780	-4	-68	-852	remote
12140	148	2740	15028	oldbin
11360	144	2672	14176	bin

-780	-4	-60	-844	saml21-xpro
11872	148	2844	14864	oldbin
11092	144	2784	14020	bin

-836	-4	-68	-908	samr21-xpro
15224	156	2740	18120	oldbin
14388	152	2672	17212	bin

-732	-4	-92	-828	spark-core
13668	148	2884	16700	oldbin
12936	144	2792	15872	bin

-780	-4	-92	-876	stm32f0discovery
14904	148	2764	17816	oldbin
14124	144	2672	16940	bin

-768	-4	-100	-872	stm32f3discovery
13004	148	2780	15932	oldbin
12236	144	2680	15060	bin

-768	-4	-92	-864	stm32f4discovery
13208	152	2768	16128	oldbin
12440	148	2676	15264	bin

-784	-4	-124	-912	udoo
12028	152	2824	15004	oldbin
11244	148	2700	14092	bin

-792	-4	-84	-880	yunjia-nrf51822
11760	148	2748	14656	oldbin
10968	144	2664	13776	bin
```

It seems that timer_init() (from the periph/timer interface) pulls in ~2kB of libgcc division code, while the (soon obsolete) hwtimer_arch implementation does not.

gebart, 2015-09-04:
I will refactor/rewrite the timer periph implementation on kinetis when I
have time. I will add support for the low power timer as well then.

gebart, 2015-10-14:
the rework in #4065 should fix this eventually.

gebart, 2016-02-16:
Fixed by #4040

closed 2016-02-16 by gebart