RIOT-OS/RIOT #3791: gnrc: Provide generic netapi ping

authmillenon, 2015-09-07:
In the spirit of #3389 and the old layer 2 ping a generic ping application/module for any `gnrc_netapi` enabled module might be of use for testing.

miri64, 2016-10-17:
@OlegHahm didn't you work on something like this?

OlegHahm, 2017-01-15:
Hm, yes, it might be, but it's probably horribly outdated.

miri64, 2019-09-10:
I think this feature could still be interesting. Let's give it another 6 months ;-).

maribu, 2023-05-17:
I'm not sure if I understand the feature request correctly. Is it ping as in ICMPv6 echo request and reply? Or is it some generic mechanism to check if network stack component is still alive and responsive? Or something else completely?

miri64, 2023-05-23:
The first: the pre-GNRC network stack had the ability to do (non-standardized) ICMPv6-echo-like behavior on the link layer. The idea was, to provide something generic based on netapi for any (packet based) layer. Not sure if this is something we still want or need though... Especially since the link layer (which started this whole discussion) is technically not part of GNRC anymore...

maribu, 2023-05-23:
OK, then lets close this and see if someone complaints :)

closed 2023-05-23 by maribu