RIOT-OS/RIOT #3794: Rpl: Invalid padding in DIS

DipSwitch, 2015-09-07:
The padding on de DIS packet is invalid according to the RFC.

Section 6.7.2 - RFC 6550:
The Pad1 option is used to insert a single octet of padding into the    message to enable options alignment.  If more than one octet of    padding is required, the PadN option should be used rather than    multiple Pad1 options.

So instead of 'memset(dis+1, 0, 4)' you must use the sequence '0x01, 0x02, 0x00, 0x00'

I'll create a PR tomorrow :)

cgundogan, 2015-09-08:
Thanks for pointing this out @DipSwitch. I agree that this should be changed to PadN. However, I wouldn't call the  use of multiple PAD1s invalid as per RFC (it says should, not must (: ). Actually, including this padding is more a workaround and only necessary for the native case, to fulfill the minimum length of eth frames. This shouldn't be included in 802.15 frames e.g.

DipSwitch, 2015-09-08:
Ah I see, it was late yesterday sorry :)

DipSwitch, 2015-09-08:
Fixed in #3796

closed 2015-09-08 by DipSwitch