RIOT-OS/RIOT #3798: xtimer: xtimer_usleep_until() fails on stm32f4discovery

haukepetersen, 2015-09-08:
I was trying to run the following on the `stm32f4discovery`:
```
#define INTERVAL (1000000U)
int main(void)
{
    uint32_t last_wakeup = xtimer_now();

    while(1) {
        xtimer_usleep_until(&last_wakeup, INTERVAL);
        LED_RED_TOGGLE;
    }

    return 0;
}
```
It is basically the `timer_periodic_wakeup` example, just taken out the printf and substituted it with an led toggle. With printf the example works as expected. But with the code shown above, the timer does not sleep for a second anymore, but the LED is blinking wild @666 KHz.

I noticed, that the first call to `xtimer_usleep_until()` does actually let the task sleep for the expected second. After this, it does not go to sleep anymore....

All xtimer tests (as xtimer_msg, xtimer_remove, etc) do work on the STM board...

On the `samr21-xpro` I was not able to reproduce this behavior -> is this faulty behavior created by the xtimer or by the STM's periph driver? 

Side note: the f4discovery runs @168MHz and is probably our fastest running board at the moment. Could this lead to any untested behavior?

haukepetersen, 2015-09-08:
On the `stm32f3discovery`, which is using almost the identical low-level timer implementation I was also not able to see the faulty behavior described above...

kaspar030, 2015-09-08:
@haukepetersen Is this on master?

Could I somehow get access to the board?

haukepetersen, 2015-09-08:
yes, it is on master. I could give you a board tonight somehow?

kaspar030, 2015-09-08:
> I could give you a board tonight somehow?

yeah, that would be good. wanna hop by after work?

> it is on master.

Can you rule out that hwtimer is interfering? (maybe, try on switch_to_xtimer branch)

PeterKietzmann, 2015-09-09:
Did #3799 fix it?

kaspar030, 2015-09-09:
It shouldn't.

haukepetersen, 2015-09-09:
nope, did not

kaspar030, 2015-09-15:
- fixed by 709a94a (in #3525)

gebart, 2015-10-14:
fix has been merged

closed 2015-10-14 by gebart