RIOT-OS/RIOT #3805: dns: Add DNS support to RIOT

DipSwitch, 2015-09-09:
Hello,

I would like to have DNS support in RIOT. I'm currently implementing the NDP RDNSS option.
But I was wondering if you would prefer RDNSS configuration trough NDP or use Multicast DNS.

For multicast DNS I would only implement the Query + Response part of the protocol,  not the node announcement etc. Although this wouldn't be to hard to implement, it would get a mess on the network I think.

Both have there own pro's and con's...

kaspar030, 2015-09-09:
> I was wondering if you would prefer RDNSS configuration trough NDP or use Multicast DNS.

definately RDNSS.

DipSwitch, 2015-09-09:
And since the DNS servers are provided via the Router Advertisement on an interface, would it be okay to add something like the following to the `gnrc_ipv6_netif_t` structure?

		#if defined(MODULE_GNRC_DNS)
			gnrc_dns_servers_t dns;
		#endif
		
Where:
		
		/**
		 * @brief   DNS server entry
		 */
		typedef struct {
			uint32_t lifetime;      /**< Lifetime of the server / OR time when entry can no longer be used */
			ipv6_addr_t dns_server[GNRC_DNS_SERVERS_SUPPORT]; /**< DNS server address */
		} gnrc_dns_servers_t;

Then the MODULE_GNRC_NDP_ROUTER can also use it to send his entries to other nodes.
But I can also make a real separate module of it.

DipSwitch, 2015-09-09:
That was a stupid idea, it will become a completely separated module =)

DipSwitch, 2015-09-11:
And what do you think about caching the DNS entries? Would eat to much resources I guess?

DipSwitch, 2015-09-12:
Closed for discussion can continue in #3823

closed 2015-09-12 by DipSwitch