RIOT-OS/RIOT #3812: core: Compile time macro byte swapping for HTONS, HTONL, NTOHS, NTOHL etc.

DipSwitch, 2015-09-10:
This would be useful so you can create enums like:

    # define CT_HTONS(x)	(((x >> 8) & 0x00FF) | ((x << 8) & 0xFF00))
    # define CT_HTONL(x)	(((x >> 24) & 0x000000FF) | ((x >> 8) & 0x0000FF00) | ((x << 8) & 0x00FF0000) | ((x << 24) & 0xFF000000))

    typedef enum {
        A                       = CT_HTONS(0x0001),
        AAAA                    = CT_HTONS(0x001C),
    } dns_query_type;

We already use this our self in some projects.

gebart, 2015-09-10:
Doesn't byteorder.h provide these macros?

DipSwitch, 2015-09-10:
Only static inline functions, these the compiler can't optimize when defining enum's.

gebart, 2016-02-09:
Good idea. 
@DipSwitch  do you want to provide a PR?

haukepetersen, 2016-03-31:
postponed to 2016.07

kYc0o, 2016-07-25:
Postponed again to 2010.10

PeterKietzmann, 2017-01-26:
How about #5453

closed 2019-09-10 by stale[bot]