RIOT-OS/RIOT #3815: Multi-Hop UDP results in a hard fault (iotlab-m3)

cgundogan, 2015-09-10:
I was testing on the iotlab testbed multi hop UDP with the following configuration:
```
m3-190;ifconfig 7 add abcd::190
m3-195;ifconfig 7 add abcd::195
m3-205;ifconfig 7 add abcd::205

m3-190;fibroute add :: via fe80::195
m3-195;fibroute add :: via fe80::205
m3-195;fibroute add abcd::190 via fe80::190
m3-205;fibroute add abcd::195 via fe80::195
m3-205;fibroute add abcd::190 via fe80::195

udp server start 1337
```
Issuing the following command:
```
m3-205;udp send abcd::190 1337 hey
```
Results in the following message:
```
m3-205;udp send abcd::190 1337 hey
1441915054.206381;m3-205;> udp send abcd::190 1337 hey
1441915054.208265;m3-205;Success: send 3 byte to abcd::190:1337
1441915054.217298;m3-190;> PKTDUMP: data received:
1441915054.217407;m3-190;~~ SNIP  0 - size:   3 byte, type: NETTYPE_UNDEF (0)
1441915054.217499;m3-190;000000 68 65 79
1441915054.218378;m3-190;~~ SNIP  1 - size:   8 byte, type: NETTYPE_UDP (4)
1441915054.219366;m3-190;   src-port:  1337  dst-port:  1337
1441915054.220344;m3-190;   length: 11  cksum: 0x4b8d0
1441915054.221388;m3-190;~~ SNIP  2 - size:  40 byte, type: NETTYPE_IPV6 (2)
1441915054.222358;m3-190;traffic class: 0x00 (ECN: 0x0, DSCP: 0x00)
1441915054.222462;m3-190;flow label: 0x00000
1441915054.223356;m3-190;length: 11  next header: 17  hop limit: 63
1441915054.225320;m3-190;source address: abcd::205
1441915054.225426;m3-190;destination address: abcd::190
1441915054.225513;m3-190;~~ SNIP  3 - size:  24 byte, type: NETTYPE_NETIF (-1)
1441915054.226082;m3-190;if_pid: 7  rssi: 18  lqi: 126
1441915054.227045;m3-190;src_l2addr: 36:32:48:33:46:de:8f:16
1441915054.227142;m3-190;dst_l2addr: 36:32:48:33:46:d8:7d:0a
1441915054.228033;m3-190;~~ PKT    -  4 snips, total size:  75 byte
1441915054.228136;m3-190;
1441915054.229034;m3-190;Context before hardfault:
1441915054.229135;m3-190;r0:  0x20001930
1441915054.229220;m3-190;r1:  0x28
1441915054.230020;m3-190;r2:  0x4902b5
1441915054.230118;m3-190;r3:  0x4902b5
1441915054.230201;m3-190;r12: 0x2000184b
1441915054.231025;m3-190;lr : 0x80043fb
1441915054.231122;m3-190;pc : 0x80040fa
1441915054.231205;m3-190;psr: 0x81000000
1441915054.231287;m3-190;
1441915054.232018;m3-190;*** RIOT kernel panic
1441915054.232117;m3-190;HARD FAULT HANDLER
1441915054.234038;m3-190;       pid | name                 | state    Q | pri | stack ( used) | location
1441915054.235016;m3-190;         1 | idle                 | pending  Q |  15 |   256 (  140) | 0x20000230
1441915054.237028;m3-190;         2 | main                 | bl mutex _ |   7 |  1536 (  708) | 0x20000330
1441915054.239011;m3-190;         3 | pktdump              | running  Q |   6 |  1536 (  556) | 0x20002f58
1441915054.241029;m3-190;         4 | 6lo                  | bl rx    _ |   3 |  1024 (  408) | 0x2000355c
1441915054.242020;m3-190;         5 | ipv6                 | bl rx    _ |   4 |  1024 (  488) | 0x20000e0c
1441915054.244045;m3-190;         6 | udp                  | bl rx    _ |   5 |  1024 (  316) | 0x20003b38
1441915054.246007;m3-190;         7 | at86rfxx             | bl rx    _ |   3 |  1024 (  360) | 0x200009cc
1441915054.247019;m3-190;           | SUM                  |            |     |  7424 ( 2976)
1441915054.247125;m3-190;
1441915054.247214;m3-190;*** halted.
```

Ping works like a charm. It's just UDP that fails AFAICT.
Can somebody with proper hardware investigate further?

cgundogan, 2015-09-10:
tests were conducted with current master

OlegHahm, 2015-09-10:
What do you mean by "proper hardware"? In my experience `ENABLE_DEBUG` is often far more efficient than `gdb` for debugging network problems.

OlegHahm, 2015-09-10:
Have you checke dthe addresses of the hardfault?

cgundogan, 2015-09-11:
>Have you checke dthe addresses of the hardfault?

Yes, sorry for omitting this piece of information:
* `pc` is pointing to `_pktbuf_free`
* `lr` to `gnrc_pktbuf_release`.

authmillenon, 2015-09-11:
Looks like a broken packet buffer :/

cgundogan, 2015-09-11:
I just tested on native => is reproducable. So this should be fairly easy to debug. I have to leave now however, will look into this later today if nobody else fixes this until then

cgundogan, 2015-09-11:
@authmillenon the segfault seems to happen here: [gnrc_udp.c](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/transport_layer/udp/gnrc_udp.c#L135)

cgundogan, 2015-09-11:
subsequent tests yield the segfault at this position: [gnrc_pktdump.c#117](https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/pktdump/gnrc_pktdump.c#L117)

authmillenon, 2015-09-11:
It all points to some use after release IMHO.

authmillenon, 2015-09-11:
Was this behavior apparent in older versions? If not: a bisect might be useful to pinpoint the commit that introduced it.

cgundogan, 2015-09-11:
`git bisect`

There are only 'skip'ped commits left to test.
The first bad commit could be any of:
3a21cefd6bce36fa26eef056c6439c343cb3981c
b4c00d5f6e439155527e307a2057610e360fd97b
8b16f3ef06bc8c98df816871420b8f425a525855
We cannot bisect more!

cgundogan, 2015-09-11:
the first both commits seem to be less intrusive. So I guess its 8b16f3e

cgundogan, 2015-09-11:
FYI: tested with #3813 merged into master - same erroneous behavior

cgundogan, 2015-09-11:
I did more multi-hop tests on the iotlab-testbed with udp traffic and this is what I encountered:
```
1441996879.169776;m3-275;> udp send abcd::3432:4833:46d4:7c16 1337 hey
1441996879.173339;m3-275;Success: send 3 byte to abcd::3432:4833:46d4:7c16:1337
1441996879.176005;m3-285;PKTDUMP: data received:
1441996879.177054;m3-285;~~ SNIP  0 - size:   3 byte, type: NETTYPE_UNDEF (0)
1441996879.177159;m3-285;000000 68 65 79
1441996879.178164;m3-285;~~ SNIP  1 - size:   8 byte, type: NETTYPE_UDP (4)
1441996879.178291;m3-285;   src-port:  1337  dst-port:  1337
1441996879.179283;m3-285;   length: 11  cksum: 0x42bb0
1441996879.180358;m3-285;~~ SNIP  2 - size:  40 byte, type: NETTYPE_IPV6 (2)
1441996879.181460;m3-285;traffic class: 0x00 (ECN: 0x0, DSCP: 0x00)
1441996879.181567;m3-285;flow label: 0x00000
1441996879.182530;m3-285;length: 11  next header: 17  hop limit: 64
1441996879.184372;m3-285;source address: abcd::3432:4833:46d9:8e26
1441996879.184479;m3-285;destination address: abcd::3432:4833:46d4:7c16
1441996879.185079;m3-285;~~ SNIP  3 - size:  24 byte, type: NETTYPE_NETIF (-1)
1441996879.185882;m3-285;if_pid: 7  rssi: 12  lqi: 84
1441996879.186865;m3-285;src_l2addr: 36:32:48:33:46:d9:8e:26
1441996879.186969;m3-285;dst_l2addr: 36:32:48:33:46:d4:7c:16
1441996879.187867;m3-285;~~ PKT    -  4 snips, total size:  75 byte
1441996883.231193;m3-285;assertion "_pktbuf_contains(pkt)" failed: file "RIOT/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c", line 203, function: gnrc_pktbuf_release
1441996883.232300;m3-285;#! exit 1: resetting
1441996883.232430;m3-275;>
1441996883.232528;m3-275;Context before hardfault:
1441996883.232618;m3-275;r0:  0x1
1441996883.233152;m3-275;r1:  0x0
1441996883.233253;m3-275;r2:  0x1
1441996883.233338;m3-275;r3:  0x8000467
1441996883.233422;m3-275;r12: 0x0
1441996883.233504;m3-275;lr : 0x80008d9
1441996883.234850;m3-275;pc : 0x80043ce
1441996883.234850;m3-275;psr: 0x21000000
1441996883.234850;m3-275;
1441996883.235054;m3-275;*** RIOT kernel panic
1441996883.235153;m3-275;HARD FAULT HANDLER
1441996883.237063;m3-275;       pid | name                 | state    Q | pri | stack ( used) | location
1441996883.238850;m3-275;         1 | idle                 | pending  Q |  15 |   256 (  140) | 0x20000230
1441996883.240187;m3-275;         2 | main                 | bl mutex _ |   7 |  1536 (  860) | 0x20000330
1441996883.242850;m3-275;         3 | pktdump              | bl rx    _ |   6 |  1536 (  244) | 0x20002f58
1441996883.244186;m3-275;         4 | 6lo                  | running  Q |   3 |  1024 (  408) | 0x2000355c
1441996883.245632;m3-275;         5 | ipv6                 | bl rx    _ |   4 |  1024 (  564) | 0x20000e0c
1441996883.248186;m3-275;         6 | udp                  | bl rx    _ |   5 |  1024 (  316) | 0x20003b38
1441996883.249614;m3-275;         7 | at86rfxx             | bl rx    _ |   3 |  1024 (  360) | 0x200009cc
1441996883.250850;m3-275;         8 | RPL                  | bl rx    _ |   3 |  1024 (  688) | 0x20005480
1441996883.252789;m3-275;           | SUM                  |            |     |  8448 ( 3580)
1441996883.252899;m3-275;
1441996883.252990;m3-275;*** halted.
1441996883.253075;m3-275;
```

Surprisingly, this time the sender (`m3-275`) did crash with a hard fault. More surprisingly, at the exact same time, the receiver of the udp traffic (`m3-285`) hit an assertion. So that in the end, both nodes were rendered unusable.

Hard fault:
* pc: `gnrc_pktbuf_hold`
* lr: `atomic_set_to_one`

OlegHahm, 2015-09-11:
Oh no!

cgundogan, 2015-09-11:
life is too boring without hard faults!

authmillenon, 2015-09-21:
@cgundogan is this still an issue?

cgundogan, 2015-09-21:
didn't do much testing lately, fighting a cold currently :cry: Will test it later in the evening

authmillenon, 2015-09-21:
Can you maybe provide a test setup, so that others can test, too? Hope you get well soon.

cgundogan, 2015-09-21:
I did the tests manually, because this way I had the best feedback about hard faults or failing connections due to interferences.
I did use the automated tests that I wrote for the release specs, but it was not possible to retrieve hard fault information in a simple way and this is not good, if the hard faults only come sporadically. So what I basically did was configuring static routes between three or more nodes that are sufficiently apart from each other (on the iot-lab testbed) and used the udp command to test if it works or not. Then I tried various combinations of who talks to whom (single and multi-hop) because I also had several cases in the past where a node in the middle starts to misbehave after several attempts. The last known bug that I encountered was a hard fault when I sent a udp packet to the node's own link-local address and afterwards one packet to another address. I could **almost** recreate this hard fault reliably, but I need to test this again today, because there were quite some changes in the network stack in the past days

cgundogan, 2015-09-21:
Here I did a simple test again, sending a UDP packet to the node's own link local address on the iotlab testbed:
```
1442846389.553721;m3-1;> udp send fe80::3432:4833:46d7:ab12 1337 hey
1442846389.553881;m3-1;PKTDUMP: data received:
1442846389.554921;m3-1;~~ SNIP  0 - size:   3 byte, type: NETTYPE_UNDEF (0)
1442846389.555030;m3-1;000000 68 65 79
1442846389.555400;m3-1;~~ SNIP  1 - size:   8 byte, type: NETTYPE_UDP (4)
1442846389.556595;m3-1;   src-port:  1337  dst-port:  1337
1442846389.556707;m3-1;   length: 11  cksum: 0x43a60
1442846389.557635;m3-1;~~ SNIP  2 - size:  40 byte, type: NETTYPE_IPV6 (2)
1442846389.558850;m3-1;traffic class: 0x00 (ECN: 0x0, DSCP: 0x00)
1442846389.560186;m3-1;flow label: 0x00000
1442846389.560896;m3-1;length: 11  next header: 17  hop limit: 64
1442846389.562851;m3-1;source address: fe80::3432:4833:46d7:ab12
1442846389.562851;m3-1;destination address: fe80::3432:4833:46d7:ab12
1442846389.562892;m3-1;~~ PKT    -  3 snips, total size:  51 byte
1442846389.564882;m3-1;Success: send 40 byte to fe80::3432:4833:46d7:ab12:1337
1442846390.302882;m3-1;
1442846390.302882;m3-1;Context before hardfault:
1442846390.304292;m3-1;r0:  0x20001cf4
1442846390.304292;m3-1;r1:  0x4
1442846390.304292;m3-1;r2:  0x44902b5
1442846390.305363;m3-1;r3:  0x44902b5
1442846390.305363;m3-1;r12: 0xff
1442846390.305363;m3-1;lr : 0x800405f
1442846390.305363;m3-1;pc : 0x8003d66
1442846390.306599;m3-1;psr: 0x81000000
1442846390.306599;m3-1;
1442846390.306851;m3-1;*** RIOT kernel panic
1442846390.306851;m3-1;HARD FAULT HANDLER
1442846390.307666;m3-1; pid | name                 | state    Q | pri | stack ( used) | location
1442846390.309942;m3-1;   1 | idle                 | pending  Q |  15 |   256 (  136) | 0x20000204
1442846390.311999;m3-1;   2 | main                 | bl mutex _ |   7 |  1536 (  716) | 0x20000304
1442846390.313016;m3-1;   3 | pktdump              | bl rx    _ |   6 |  1536 (  556) | 0x20003440
1442846390.314892;m3-1;   4 | 6lo                  | bl rx    _ |   3 |  1024 (  408) | 0x20003a44
1442846390.317010;m3-1;   5 | ipv6                 | pending  Q |   4 |  1024 (  400) | 0x20000dd4
1442846390.318002;m3-1;   6 | udp                  | bl rx    _ |   5 |  1024 (  320) | 0x200041ec
1442846390.320658;m3-1;   7 | at86rfxx             | running  Q |   3 |  1024 (  360) | 0x20000994
1442846390.321840;m3-1;     | SUM                  |            |     |  7424 ( 2896)
1442846390.321840;m3-1;
1442846390.321840;m3-1;*** halted.

```

cgundogan, 2015-09-21:
The crash does not come immediately, when I tried to debug this during the weekend I found out that it crashes as soon as the node wants to send a RA. But I don't think that it's related to the RA, I just think the packet buffer gets corrupted somewhere else

authmillenon, 2015-09-21:
Did you test this now, or during the weekend? Because there were some fixes for router advertisements this weekend.

cgundogan, 2015-09-21:
Now and with current master

cgundogan, 2015-09-21:
Oh let me test this with the RC branch...wait a sec

cgundogan, 2015-09-21:
same behavior with the @OlegHahm's RC branch. Crash after a udp packet to the node's own link-local address. Note that the crash does not come immediately, but after the node is attempting to send the RA.

authmillenon, 2015-09-21:
Will investigate.

OlegHahm, 2015-09-21:
Probably a timer being null again. Have you tried to add some asserts into xtimer (most important to `xtimer_remove()` because this is called in most places before the actual xtimer_set... call because of vtimer legacis.

authmillenon, 2015-09-21:
No it seems to be purely UDP related this time, as far as I can see at least. I checked the packet buffer using GDB. When pinging a node it is empty afterwards, when sending UDP packets there is still something in it which looks broken (`_first_unused == _pktbuf`, `*_first_unused: <next=_pktbuf+188, size=192>` => the unused chunk goes into the next unused chunk, since 188 < 192)

authmillenon, 2015-09-21:
Will investigate further.

cgundogan, 2015-09-21:
When a pktsnip is send down the stack and it notices that it's a packet for the node itself, then the packet is sent right back to the application, am I right? Does any layer in between generate a copy of this pktsnip, or is it the same pointer that was passed down? Let's fix this before microsoft fixes skype!

cgundogan, 2015-09-21:
BTW, I tested again sending a UDP packet to the same node but **without** starting the udp server. Still crashes right before the RA goes out. So the bug may not be on the receiving side of UDP (but in lower layers)

cgundogan, 2015-09-21:
I may have found the root of all evil: it's the pktbuffer :P But this surpasses my knowledge so I hand this over to you @authmillenon.
The problem only occures if the udp payload size is very small. There is no hard fault if you choose a succifiently big payload for udp. I think the problem is in the logic of `_pktbuf_free` .. if the payload is too small, it gets not merged, but somehow resides in the pktbuffer and lets the next packet (which is the periodic RA) crash.

authmillenon, 2015-09-21:
I'll try to investigate tomorrow. Just for certainty: can you maybe test ping with very short payloads also?

cgundogan, 2015-09-21:
This won't work because the payload of ping and the icmp header reside in the same pktsnip. So the pktsnip is sufficiently big (IIRC). But in udp, the payload resides in its own pktsnip with type UNDEF

authmillenon, 2015-09-22:
Was able to reproduce your hypothesis in unittests:

```C
static void test_pktbuf_release__short_pktsnips(void)
{
    gnrc_pktsnip_t *pkt = gnrc_pktbuf_add(NULL, TEST_STRING8, sizeof(TEST_STRING8),
                                          GNRC_NETTYPE_UNDEF);
    gnrc_pktsnip_t *hdr = gnrc_pktbuf_mark(pkt, sizeof(TEST_STRING8) - 1, GNRC_NETTYPE_TEST);
    TEST_ASSERT(pkt);
    TEST_ASSERT(hdr);
    TEST_ASSERT(pkt->next == hdr);
    TEST_ASSERT(hdr->next == NULL);
    gnrc_pktbuf_release(pkt);
    TEST_ASSERT(gnrc_pktbuf_is_empty());
}
```

```
main(): This is RIOT! (Version: 2014.12-3529-g00a0-beutlin)
................................
gnrc_pktbuf_tests.test_pktbuf_release__short_pktsnips (tests/unittests/tests-pktbuf/tests-pktbuf.c 587) gnrc_pktbuf_is_empty()
.......
run 39 failures 1
```

authmillenon, 2015-09-22:
Will debug.

authmillenon, 2015-09-22:
Fixed in #3925. With the PR included:

```
1442912001.053657;m3-9;> udp server start 1337
1442912001.054507;m3-9;Success: started UDP server on port 1337
udp send fe80::3432:4833:46d5:9d12 1337 hey
1442912013.195618;m3-9;> udp send fe80::3432:4833:46d5:9d12 1337 hey
1442912013.196562;m3-9;PKTDUMP: data received:
1442912013.197542;m3-9;~~ SNIP  0 - size:   3 byte, type: NETTYPE_UNDEF (0)
1442912013.197644;m3-9;000000 68 65 79
1442912013.198543;m3-9;~~ SNIP  1 - size:   8 byte, type: NETTYPE_UDP (4)
1442912013.199592;m3-9;   src-port:  1337  dst-port:  1337
1442912013.200480;m3-9;   length: 11  cksum: 0x45664
1442912013.201596;m3-9;~~ SNIP  2 - size:  40 byte, type: NETTYPE_IPV6 (2)
1442912013.202565;m3-9;traffic class: 0x00 (ECN: 0x0, DSCP: 0x00)
1442912013.202671;m3-9;flow label: 0x00000
1442912013.203566;m3-9;length: 11  next header: 17  hop limit: 64
1442912013.204574;m3-9;source address: fe80::3432:4833:46d5:9d12
1442912013.205561;m3-9;destination address: fe80::3432:4833:46d5:9d12
1442912013.206558;m3-9;~~ PKT    -  3 snips, total size:  51 byte
1442912013.207561;m3-9;Success: send 40 byte to [fe80::3432:4833:46d5:9d12]:1337
```

```
(gdb) c
Continuing.
^C^[[A
Program received signal SIGINT, Interrupt.
idle_thread (arg=<optimized out>) at /home/martine/Repositories/RIOT-OS/RIOT/core/kernel_init.c:73
73	        if (lpm_prevent_sleep) {
(gdb) print _first_unused
$4 = (_unused_t *) 0x20001c38 <_pktbuf>
(gdb) print *_first_unused
$5 = {next = 0x0, size = 6144}
```

(This means the packet buffer is empty after UDP packet receival). And no more hard faults

PR #3821 (2015-09-11): gnrc_ndp: add PIO to router advertisements
The PIOs for the configured prefixes were generated, but never added to
the packet, causing some problems in the packet buffer.

Fixes #3815 (possibly)
commit e5178691 (2015-09-11): gnrc_ndp: add PIO to router advertisements

The PIOs for the configured prefixes were generated, but never added to
the packet, causing some problems in the packet buffer.

Fixes #3815 (possibly)
PR #3925 (2015-09-22): gnrc_pktbuf_static: fix marking of pktsnips with short payload
Fixes #3815 and provides a test for the situation in which the bug occurs: If a payload of a packet is smaller then the `_unused_t` marker of the packet buffer the marker overrides the following data when the packet is released. Due to that such packets needs to be moved on marking.
closed 2015-09-22 by authmillenon