RIOT-OS/RIOT #3817: ndp: RIOT node assignes wrong IP when receiving RTR_ADV PI option

DipSwitch, 2015-09-11:
From RFC https://tools.ietf.org/html/rfc4861#section-4.6.2 :

      Prefix         An IP address or a prefix of an IP address.  The
                     Prefix Length field contains the number of valid
                     leading bits in the prefix.  The bits in the prefix
                     after the prefix length are reserved and MUST be
                     initialized to zero by the sender and ignored by
                     the receiver.  A router SHOULD NOT send a prefix
                     option for the link-local prefix and a host SHOULD
                     ignore such a prefix option.

If i'm correct that would mean that the although the `radvd` daemon on Linux sends an none advised  PI option. RIOT parses the option in a none advised way.

However `radvd` sends the following packet now:

          prefix info option (3), length 32 (4): fdcb:172:32::/64, Flags [onlink, auto], valid time 86400s, pref. time 14400s
            0x0000:  40c0 0001 5180 0000 3840 0000 0000 fdcb
            0x0010:  0172 0032 0000 0000 0000 0000 0000
          rdnss option (25), length 24 (3):  lifetime 900s, addr: fdcb:172:31::1:254
            0x0000:  0000 0000 0384 fdcb 0172 0031 0000 0000
            0x0010:  0000 0001 0254

This results in the following IP assignment:

    Iface  8   HWaddr: 87:36  Channel: 5  NID: 0x23  TX-Power: 0dBm  State: IDLE
               Long HWaddr: 30:34:35:37:42:bb:87:36  CSMA Retries: 4
               AUTOACK  CSMA  MTU:1280  6LO  IPHC
               inet6 addr: fe80::3234:3537:42bb:8736/64  scope: local
               inet6 addr: fdcb:172:32::/64  scope: global

I would assume that the assigned IP should be something like:

           inet6 addr: fdcb:172:32::3234:3537:42bb:8736/64  scope: global

I'm trying to debug this, but I can't find how to fix this so fast...

Another problem is that when `radvd` sends:

          prefix info option (3), length 32 (4): fdcb:172:32::1/64, Flags [onlink, auto], valid time 86400s, pref. time 14400s
            0x0000:  40c0 0001 5180 0000 3840 0000 0000 fdcb
            0x0010:  0172 0032 0000 0000 0000 0000 0001

The assigned IP would be:

               inet6 addr: fdcb:172:32::1/64  scope: global

authmillenon, 2015-09-11:
Got a fix already

DipSwitch, 2015-09-11:
Ah awesome! :D Where to pull? :)

authmillenon, 2015-09-11:
#3818

DipSwitch, 2015-09-11:
Works like a charm! :+1:
Thx

PR #3818 (2015-09-11): gnrc_ndp: fix setting of addresses by PIO
Fixes #3817
closed 2015-09-11 by DipSwitch