RIOT-OS/RIOT #3824: native: gnrc: hardcore pinging crashes

OlegHahm, 2015-09-11:
Sending 1.000.000 pings between two native instances in gnrc_networking example with a payload of 1024 bytes and a delay of 0 crashes the sending node after several 100.000 pings. Reducing the payload to something smaller (e.g. 512 bytes) works fine.

cgundogan, 2015-09-11:
When I use ping6 with a payload 1024 it just times out. Does it really crash for you?
```
> ping6 2 abcd::2 10 0
ping6 2 abcd::2 10 0
18 bytes from abcd::2: id=92 seq=1 hop limit=64 time = 0.676 ms
18 bytes from abcd::2: id=92 seq=2 hop limit=64 time = 0.481 ms
--- abcd::2 ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 0.061254 s
rtt min/avg/max = 0.481/0.578/0.676 ms
> ping6 2 abcd::2 1024 0
ping6 2 abcd::2 1024 0
ping timeout
ping timeout
--- abcd::2 ping statistics ---
2 packets transmitted, 0 received, 100% packet loss
```

OlegHahm, 2015-09-11:
@cgundogan, yes, if you merge #3826.

cgundogan, 2015-09-11:
>@cgundogan, yes, if you merge #3826.

I was missing that piece of info in the description ;)
After merging #3826: I can confirm that the pinging native node crashes after a huge amount pings.
```
Program received signal SIGTRAP, Trace/breakpoint trap.
0x00000000 in ?? ()
(gdb) bt
#0  0x00000000 in ?? ()
#1  0x080831a7 in _netdev2_eth_stack ()
#2  0xf7dd423b in setcontext () from /usr/lib32/libc.so.6
#3  0x0804ca33 in isr_cpu_switch_context_exit () at RIOT/cpu/native/native_cpu.c:141
#4  0x080865a0 in _netdev2_eth_stack ()
#5  0x21e2b100 in ?? ()
#6  0x00000005 in ?? ()
#7  0x0807dd8c in __isr_stack ()
#8  0x0808313b in _netdev2_eth_stack ()
#9  0x21e2b100 in ?? ()
#10 0x08082500 in ?? ()
Backtrace stopped: Cannot access memory at address 0x12340025
```

OlegHahm, 2015-09-11:
Looked similar here.

authmillenon, 2015-09-22:
@OlegHahm is this still true with the patch provided in #3925?

authmillenon, 2015-09-22:
I still get crasches so yes. But the `pktbuf` does not look suspicious (There is something in it, but that's to be expected when it crasches during "hardcore pinging"):

```
Program received signal SIGTRAP, Trace/breakpoint trap.
0x00000000 in ?? ()
(gdb) print _first_unused 
$1 = (_unused_t *) 0x8089170 <_pktbuf+80>
(gdb) print *_first_unused 
$2 = {next = 0x80899c8 <_pktbuf+2216>, size = 1064}
(gdb) print *_first_unused->next
$3 = {next = 0x0, size = 3928}
(gdb) x/80xb _pktbuf
0x8089120 <_pktbuf>:	0x02	0x00	0x00	0x00	0x5c	0x91	0x08	0x08
0x8089128 <_pktbuf+8>:	0xc0	0x95	0x08	0x08	0x08	0x04	0x00	0x00
0x8089130 <_pktbuf+16>:	0x02	0x00	0x00	0x00	0x02	0x00	0x00	0x00
0x8089138 <_pktbuf+24>:	0x00	0x00	0x00	0x00	0x48	0x91	0x08	0x08
0x8089140 <_pktbuf+32>:	0x14	0x00	0x00	0x00	0xff	0xff	0xff	0xff
0x8089148 <_pktbuf+40>:	0x06	0x06	0x06	0x00	0x00	0x00	0x00	0x00
0x8089150 <_pktbuf+48>:	0xf2	0x6c	0x0e	0x3a	0xde	0xb9	0x3e	0x0f
0x8089158 <_pktbuf+56>:	0x06	0xd0	0x61	0x88	0x02	0x00	0x00	0x00
0x8089160 <_pktbuf+64>:	0x34	0x91	0x08	0x08	0x98	0x95	0x08	0x08
0x8089168 <_pktbuf+72>:	0x28	0x00	0x00	0x00	0x01	0x00	0x00	0x00
(gdb) print 2216-1064
$4 = 1152
(gdb) x/1152xb _pktbuf
0x8089120 <_pktbuf>:	0x02	0x00	0x00	0x00	0x5c	0x91	0x08	0x08
0x8089128 <_pktbuf+8>:	0xc0	0x95	0x08	0x08	0x08	0x04	0x00	0x00
0x8089130 <_pktbuf+16>:	0x02	0x00	0x00	0x00	0x02	0x00	0x00	0x00
0x8089138 <_pktbuf+24>:	0x00	0x00	0x00	0x00	0x48	0x91	0x08	0x08
0x8089140 <_pktbuf+32>:	0x14	0x00	0x00	0x00	0xff	0xff	0xff	0xff
0x8089148 <_pktbuf+40>:	0x06	0x06	0x06	0x00	0x00	0x00	0x00	0x00
0x8089150 <_pktbuf+48>:	0xf2	0x6c	0x0e	0x3a	0xde	0xb9	0x3e	0x0f
0x8089158 <_pktbuf+56>:	0x06	0xd0	0x61	0x88	0x02	0x00	0x00	0x00
0x8089160 <_pktbuf+64>:	0x34	0x91	0x08	0x08	0x98	0x95	0x08	0x08
0x8089168 <_pktbuf+72>:	0x28	0x00	0x00	0x00	0x01	0x00	0x00	0x00
0x8089170 <_pktbuf+80>:	0xc8	0x99	0x08	0x08	0x28	0x04	0x00	0x00
0x8089178 <_pktbuf+88>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089180 <_pktbuf+96>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089188 <_pktbuf+104>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089190 <_pktbuf+112>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089198 <_pktbuf+120>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891a0 <_pktbuf+128>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891a8 <_pktbuf+136>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891b0 <_pktbuf+144>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891b8 <_pktbuf+152>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891c0 <_pktbuf+160>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891c8 <_pktbuf+168>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891d0 <_pktbuf+176>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891d8 <_pktbuf+184>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891e0 <_pktbuf+192>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891e8 <_pktbuf+200>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891f0 <_pktbuf+208>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x80891f8 <_pktbuf+216>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089200 <_pktbuf+224>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089208 <_pktbuf+232>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089210 <_pktbuf+240>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089218 <_pktbuf+248>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089220 <_pktbuf+256>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089228 <_pktbuf+264>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089230 <_pktbuf+272>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089238 <_pktbuf+280>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089240 <_pktbuf+288>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089248 <_pktbuf+296>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089250 <_pktbuf+304>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089258 <_pktbuf+312>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089260 <_pktbuf+320>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089268 <_pktbuf+328>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089270 <_pktbuf+336>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089278 <_pktbuf+344>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
0x8089280 <_pktbuf+352>:	0x53	0x53	0x53	0x53	0x53	0x53	0x53	0x53
---Type <return> to continue, or q <return> to quit---q
Quit
(gdb) 
```

Looks more or less like normal packet data to me, so I would safely assume that it is not a packet buffer problem.

authmillenon, 2015-09-22:
But the crashes look awefully similar to the ones in #3341. Are you sure this issue isn't a duplicate?

OlegHahm, 2015-09-22:
No, but I cannot check because I cannot even ping native from Linux on master.

authmillenon, 2015-09-22:
```bash
sudo ping6 -f -I tap0 fe80::3c0f:6ff:fed0:6188
```

is not working?

OlegHahm, 2015-09-22:
Yes, only if I https://github.com/RIOT-OS/RIOT/pull/3931

authmillenon, 2015-09-22:
(the IPv6 address being native's IPv6 address)

authmillenon, 2015-09-22:
what? There should be no 6Lo-ND on native....

OlegHahm, 2015-09-22:
Yes, but with #3931 it wouldn't reply to NBR SOL anymore.

authmillenon, 2015-09-22:
yes, because `iface` is given in that case. See #3935 for reversion.

OlegHahm, 2015-09-22:
Yes, with #3935 (even thought it's also not the right solution) this is working again. However, I cannot confirm (or deny) that this bug is caused by the same bug as in #3341.

kaspar030, 2015-12-07:
We'll probably not fix this for this release, so I'll retag.

OlegHahm, 2016-07-26:
Cannot reproduce any more.

miri64, 2016-07-26:
So can we close?

miri64, 2016-07-26:
(me neither btw)

OlegHahm, 2016-07-26:
Tested on two machines several times. Haven't seen a single crash -> close.

closed 2016-07-26 by OlegHahm

miri64, 2016-07-26:
(to be fair: two very similar machines, but maybe someone with an Ubuntu PC that isn't a ThinkPad likes to test too ^^)

OlegHahm, 2016-07-26:
I mean I tested on two machines (Lenovo Thinkpad X220 with Intel(R) Core(TM) i5-2520M CPU @ 2.50GHz and Dell Desktop machine with Intel(R) Core(TM) i7-3770 CPU @ 3.40GHz). But yes, both machines running Arch Linux with Linux 4.6.3.

OlegHahm, 2016-07-26:
Tested on a Intel(R) Core(TM) i7-2600 CPU @ 3.40GHz with Debian and Linux 4.4.0-1-amd64, too. Works also without problems.

miri64, 2016-07-26:
Ah I thought you were refering to our two machines :-)

kaspar030, 2016-07-26:
did you try multiple parallel 'ping -f'?

https://github.com/RIOT-OS/RIOT/issues/3824#issuecomment-235383107

OlegHahm, 2016-07-27:
Yes. Different issue.

OlegHahm, 2016-07-27:
https://github.com/RIOT-OS/RIOT/issues/3341