RIOT-OS/RIOT #3832: tftp: client / server implementation

DipSwitch, 2015-09-12:
Goodmidday,

We would like to implement a TFTP client and/or server in our firmware. Would you like to have it as a module under `sys/net/gnrc/application_layer/tftp*` Or should I implement it as a package?

OlegHahm, 2015-09-13:
Wouldn't we need a filesystem first?

kaspar030, 2015-09-13:
why? tftp could be used to download the system's current flash contents, or to store a to-be-flashed new image in memory.

OlegHahm, 2015-09-13:
Well, I was wondering about the 'f' in the name ;-) and about the question if this is the smartest protocol to exchange software update without ensuring anything (no security, no end-to-end-acknowledgements, no checksum...).

DipSwitch, 2015-09-14:
It has end-to-end acknowledgement, we always use CRC32 'signing' of the bin file.

DipSwitch, 2015-09-14:
But I can look for another way to update, I like things to be secure aswell and havn't really thought about that part because we use a dedicated private network. But thats not the default point of view for 6lowpan

kaspar030, 2015-09-14:
Well, tffp has one thing speaking for it: it is super simple. And actually I think security can easily be applied on top of it. So @DipSwitch, if you feel like implementing tftp, go ahead!

DipSwitch, 2015-12-01:
Closed in favor of #3911

closed 2015-12-01 by DipSwitch