RIOT-OS/RIOT #3857: Cannot make chronos

jeerakiet, 2015-09-15:
After I typed command "make", I got some errors show as below.
[image]

How can I deal with this problem ?

DipSwitch, 2015-09-16:
I think the header files don't have #ifndef _HEADER #define HEADER

kaspar030, 2015-09-16:
Hey,

On 09/15/15 17:07, jeerakiet wrote:
> After I typed command "make", I got some errors show as below.
> image
> <https://cloud.githubusercontent.com/assets/14296135/9880419/b98fe034-5bf5-11e5-9c14-3bf425a5212d.png>
> 
> How can I deal with this problem ?

I've just tried building chronos on master and it worked fine.

Which toolchain are you using?

Kaspar

jeerakiet, 2015-09-16:
I'm using GCC toolchain for MSP430.
Which toolchain are you using ? Kaspar

jeerakiet, 2015-09-16:
Thank you very much. I can do it :)

LudwigOrtmann, 2015-09-26:
@jeerakiet @DipSwitch issue resolved? fix available?

DipSwitch, 2015-12-01:
I don't know if the issue is resolved.

OlegHahm, 2015-12-01:
Well, since the issue opener didn't come back to us, I would consider this issue as solved.

closed 2015-12-01 by OlegHahm