RIOT-OS/RIOT #3863: libcoap: libcoap is broken with clang

authmillenon, 2015-09-16:
On current master (when compiling with `clang`)

```
make: Entering directory `/home/martine/Repositories/RIOT-OS/RIOT/tests/coap'
Building application "coap" for "native" with MCU "native".

"make" -C /home/martine/Repositories/RIOT-OS/RIOT/pkg/libcoap
"make" -C /home/martine/Repositories/RIOT-OS/RIOT/pkg/libcoap/libcoap
/home/martine/Repositories/RIOT-OS/RIOT/pkg/libcoap/libcoap/net.c:826:1: error: unused function 'check_opt_size'
      [-Werror,-Wunused-function]
check_opt_size(coap_opt_t *opt, unsigned char *maxpos) {
^
1 error generated.
make[2]: *** [/home/martine/Repositories/RIOT-OS/RIOT/tests/coap/bin/native/libcoap/net.o] Error 1
make[1]: *** [all] Error 2
make: *** [/home/martine/Repositories/RIOT-OS/RIOT/tests/coap/bin/native/libcoap.a] Error 2
make: Leaving directory `/home/martine/Repositories/RIOT-OS/RIOT/tests/coap'
```

cgundogan, 2015-09-16:
When I compile the coap test with `(BOARD=[(iotlab-m3)|(samr21-xpro)])? make -B clean all`, everything is fine. How can I reproduce?

cgundogan, 2015-09-16:
ah, when i switch the toolchain to clang, i get the same error

cgundogan, 2015-09-16:
#3869 fixes the described problem. However, compiling with clang, there is a subsequent error:
```
"make" -C RIOT/cpu/cortexm_common
RIOT/cpu/cortexm_common/vectors_cortexm.c:198:5: error: non-ASM statement in naked function is not supported
    core_panic(PANIC_HARD_FAULT, "HARD FAULT HANDLER");
    ^
RIOT/cpu/cortexm_common/include/vectors_cortexm.h:65:46: note: attribute is here
void hard_fault_default(void) __attribute__((naked));
                                             ^
1 error generated.
```

gebart, 2016-03-04:
>#3869 fixes the described problem. However, compiling with clang, there is a subsequent error:
>```
"make" -C RIOT/cpu/cortexm_common
RIOT/cpu/cortexm_common/vectors_cortexm.c:198:5: error: non-ASM statement in naked function is not supported
    core_panic(PANIC_HARD_FAULT, "HARD FAULT HANDLER");
    ^
RIOT/cpu/cortexm_common/include/vectors_cortexm.h:65:46: note: attribute is here
void hard_fault_default(void) __attribute__((naked));
                                             ^
1 error generated.
```

was fixed by #4134

closed 2016-03-04 by gebart