RIOT-OS/RIOT #3877: msp430: Build with Clang

gebart, 2015-09-17:
Just putting this idea out there. Clang has experimental support for MSP430. Similar to what has been done on the Cortex platforms with the TOOLCHAIN={gnu,llvm} variable could be implemented for the msp430 platforms.

Some links of interest:

- http://llvm.org/releases/3.7.0/tools/clang/docs/UsersManual.html#other-platforms
- http://blitzfunk.com/2014/03/28/facepalm-into-clang-llvm-and-llvm/

bapclenet, 2016-02-12:
@authmillenon, I don't use this board so I won't be able to review this issue

gebart, 2016-02-12:
Removing milestone since this is a really low prio issue

authmillenon, 2016-02-12:
Neverless someone(TM) should be assigned to an issue. I assign @OlegHahm temporarily, maybe he knows someone who can do this.

closed 2019-09-10 by stale[bot]