RIOT-OS/RIOT #3891: 6lowpan nd: neighbor cache entries stayed stale and timed out

OlegHahm, 2015-09-17:
When playing around with a 6LBR and a 6LR on https://github.com/OlegHahm/RIOT/tree/2015.09-RC0 I realized that all entries in both nodes' neighbor caches stayed marked as `STALE` and disappeared from the cache completely after some time. Haven't had the time to investigate further.

authmillenon, 2015-09-17:
That they just disappear is weird. Sure they did not PROBE first?

OlegHahm, 2015-09-17:
No, I'm not sure, but they were `STALE` right from the beginning, which shouldn't also not be the case, right?

authmillenon, 2015-09-17:
For the STALE I've got an idea and am already on it, but the disappearance is still weird...

OlegHahm, 2015-09-17:
Sorry, I haven't captured any PCAP. :-(

authmillenon, 2015-09-17:
#3892 fixes 6LoWPAN-ND at least that the entries change to DELAY (and then hopefully to PROBE and then to REACHABLE), but that the 6LR does not has a REACHABLE state is strange. It should be set to it as soon as it reaches https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c#L294 on NA reception.

authmillenon, 2015-09-17:
(maybe the S flag is not set for some reason in the NA?)

authmillenon, 2015-09-17:
It's to late today to investigate this. Will do tomorrow.

authmillenon, 2015-09-18:
Ah, of course they are stale. They are set by the NS not by the NA. #3892 is the full fix then (not for the disappearance though... that's still strange and we have to look into it, but my fix makes that harder, since entries are more seldom STALE now ^^)

closed 2015-09-19 by authmillenon

OlegHahm, 2015-09-19:
With current state I could indeed neither see NCEs being STALE when they shouldn't be. Also saw no more NCEs disappearing, but have to test more and longer to verify that.

PR #3892 (2015-09-19): gnrc_sixlowpan_nd: activate NC state machine
Fixes #3891 ~~partly~~.