RIOT-OS/RIOT #3933: 6LoWPAN: NCE stay UNREACHABLE

OlegHahm, 2015-09-22:
Apparently, NCEs entering `UNREACHABLE` state (e.g. if a node is disconected for some time), it will never leave this state again.

authmillenon, 2015-09-22:
So https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c#L79-L81 is never executed?

cgundogan, 2015-09-22:
@authmillenon could this be related to my empty `ncache` table while testing?

authmillenon, 2015-09-22:
Well it wasn't empty when I was testing for #3928... So this seems to be something newer.

authmillenon, 2015-09-22:
UNREACHABLE state would also just print out the nce as UNREACHABLE, not ignore it ;-)

OlegHahm, 2015-09-22:
> So https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c#L79-L81 is never executed?

Apparently not. Btw I observed this on 6LBR.

authmillenon, 2015-09-22:
Mhhhh... for me it works in current master:

```
ncache
2015-09-22 18:27:26,231 - INFO # > ncache
2015-09-22 18:27:26,237 - INFO # IPv6 address                    if  L2 address                state       type
2015-09-22 18:27:26,245 - INFO # ------------------------------------------------------------------------------
2015-09-22 18:27:26,252 - INFO # fe80::5855:445c:b94c:37ba        5  5a:55:44:5c:b9:4c:37:ba   STALE       REG
2015-09-22 18:27:26,258 - INFO # fe80::5855:686c:2535:65b2        5  5a:55:68:6c:25:35:65:b2   STALE       REG
ping6 fe80::5855:445c:b94c:37ba
[pulled cable of fe80::5855:445c:b94c:37ba]
2015-09-22 18:27:56,904 - INFO # > ping6 fe80::5855:445c:b94c:37ba
2015-09-22 18:28:01,931 - INFO # ping timeout
2015-09-22 18:28:07,980 - INFO # ping timeout
2015-09-22 18:28:14,010 - INFO # ping timeout
2015-09-22 18:28:14,014 - INFO # --- fe80::5855:445c:b94c:37ba ping statistics ---
2015-09-22 18:28:14,019 - INFO # 3 packets transmitted, 0 received, 100% packet loss
ncache
2015-09-22 18:28:16,312 - INFO # > ncache
2015-09-22 18:28:16,320 - INFO # IPv6 address                    if  L2 address                state       type
2015-09-22 18:28:16,326 - INFO # ------------------------------------------------------------------------------
2015-09-22 18:28:16,343 - INFO # fe80::5855:445c:b94c:37ba        5  5a:55:44:5c:b9:4c:37:ba   UNREACHABLE REG
2015-09-22 18:28:16,348 - INFO # fe80::5855:686c:2535:65b2        5  5a:55:68:6c:25:35:65:b2   STALE       REG
[plugged fe80::5855:445c:b94c:37ba in again]
2015-09-22 18:28:21,160 - INFO # > stale nc: 5, ipaddr: fe80::5855:445c:b94c:37ba, l2addr: 5a:55:44:5c:b9:4c:37:ba, l2addr_len: 8
2015-09-22 18:28:21,163 - INFO # nc_entry: 0x20001e58
2015-09-22 18:28:21,177 - INFO # unreachable, setting stale
ncache
2015-09-22 18:28:25,974 - INFO # ncache
2015-09-22 18:28:25,981 - INFO # IPv6 address                    if  L2 address                state       type
2015-09-22 18:28:25,998 - INFO # ------------------------------------------------------------------------------
2015-09-22 18:28:26,011 - INFO # fe80::5855:445c:b94c:37ba        5  5a:55:44:5c:b9:4c:37:ba   DELAY       REG
2015-09-22 18:28:26,016 - INFO # fe80::5855:686c:2535:65b2        5  5a:55:68:6c:25:35:65:b2   STALE       REG
ncache
2015-09-22 18:28:31,012 - INFO # > ncache
2015-09-22 18:28:31,019 - INFO # IPv6 address                    if  L2 address                state       type
2015-09-22 18:28:31,027 - INFO # ------------------------------------------------------------------------------
2015-09-22 18:28:31,032 - INFO # fe80::5855:445c:b94c:37ba        5  5a:55:44:5c:b9:4c:37:ba   REACHABLE   REG
2015-09-22 18:28:31,039 - INFO # fe80::5855:686c:2535:65b2        5  5a:55:68:6c:25:35:65:b2   STALE       REG
```

only change I made was outputting some stuff

```diff
diff --git a/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c b/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c
index b565e66..9b902f3 100644
--- a/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c
+++ b/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c
@@ -51,8 +51,15 @@ static char addr_str[IPV6_ADDR_MAX_STR_LEN];
 static void _stale_nc(kernel_pid_t iface, ipv6_addr_t *ipaddr, uint8_t *l2addr,
                       int l2addr_len)
 {
+    char lladdr_str[8 * 3];
+    char addr_str[IPV6_ADDR_MAX_STR_LEN];
+    printf("stale nc: %" PRIkernel_pid ", ipaddr: %s, l2addr: %s, l2addr_len: %d\n",
+           iface, ipv6_addr_to_str(addr_str, ipaddr, sizeof(addr_str)),
+           (l2addr_len > 0) ? gnrc_netif_addr_to_str(lladdr_str, sizeof(lladdr_str), l2addr, l2addr_len)
+           : "-", l2addr_len);
     if (l2addr_len != -ENOTSUP) {
         gnrc_ipv6_nc_t *nc_entry = gnrc_ipv6_nc_get(iface, ipaddr);
+        printf("nc_entry: %p\n", (void *)nc_entry);
         if (nc_entry == NULL) {
 #ifdef MODULE_GNRC_SIXLOWPAN_ND_ROUTER
             /* tentative type see https://tools.ietf.org/html/rfc6775#section-6.3 */
@@ -77,11 +84,13 @@ static void _stale_nc(kernel_pid_t iface, ipv6_addr_t *ipaddr, uint8_t *l2addr,
 #ifdef MODULE_GNRC_SIXLOWPAN_ND_ROUTER
         /* unreachable set in gnrc_ndp_retrans_nbr_sol() will just be staled */
         else if (gnrc_ipv6_nc_get_state(nc_entry) == GNRC_IPV6_NC_STATE_UNREACHABLE) {
+            printf("unreachable, setting stale\n");
             gnrc_ndp_internal_set_state(nc_entry, GNRC_IPV6_NC_STATE_STALE);
         }
 #endif
         else if (((uint16_t)l2addr_len != nc_entry->l2_addr_len) ||
                  (memcmp(l2addr, nc_entry->l2_addr, l2addr_len) != 0)) {
+            printf("new link layer address\n");
             /* if entry exists but l2 address differs: set */
             nc_entry->l2_addr_len = (uint16_t)l2addr_len;
             memcpy(nc_entry->l2_addr, l2addr, l2addr_len);
```

authmillenon, 2015-09-22:
(tested on `samr21-xpro`)

authmillenon, 2015-09-22:
Oh it wasn't current master, but 8df17de. But the only thing that was merged to master since then was #3932 and this was merged after this issue was opened.

authmillenon, 2015-09-22:
(pinging node was a border router)

authmillenon, 2015-09-22:
(with RPL compiled in, but since I did not activate it it shouldn't make a difference)

OlegHahm, 2015-09-22:
Interesting. My setup is:
Linux:
 * tun if: 1515::1
6LBR:
* 6lo if: 1616::1
* SLIP if: 15::2
6LR:
 * 6lo if: 1616::5

If I call `ping6 1616::5` on Linux, everything works. Now I disconnect the 6LR for some time, pinging stales (as expected) and ncache on 6LBR shows:
```
2015-09-22 20:02:40,031 - INFO # IPv6 address                    if  L2 address                state       type
2015-09-22 20:02:40,038 - INFO # ------------------------------------------------------------------------------
2015-09-22 20:02:40,045 - INFO # fe80::3432:4833:46dd:7c2a        5  36:32:48:33:46:dd:7c:2a   REACHABLE   REG
2015-09-22 20:02:40,052 - INFO # 1616::5                          5  36:32:48:33:46:dd:7c:2a   UNREACHABLE REG
2015-09-22 20:02:40,058 - INFO # 15::1                            6                            UNMANAGED   
```
(also as expected).
Now, I reconnect the 6LR and pinging resumes (as expected). However, ncache on 6LBR is still showing the same output. BUT calling `ping6 1616::5 from LBR works also fine and changes the ncache result to
```
2015-09-22 20:02:48,431 - INFO # IPv6 address                    if  L2 address                state       type
2015-09-22 20:02:48,438 - INFO # ------------------------------------------------------------------------------
2015-09-22 20:02:48,445 - INFO # fe80::3432:4833:46dd:7c2a        5  36:32:48:33:46:dd:7c:2a   DELAY       REG
2015-09-22 20:02:48,451 - INFO # 1616::5                          5  36:32:48:33:46:dd:7c:2a   REACHABLE   REG
2015-09-22 20:02:48,458 - INFO # 15::1   
```

OlegHahm, 2015-09-22:
The "good" news: the problem seem to happen already for regular forwarding without an error case.

authmillenon, 2015-09-22:
Can you give me a hint how I can set this up so I can help you find that bug?

OlegHahm, 2015-09-22:
I haven't checked if it's happening only on a 6LBR, but in general the setup is pinging over two hops. After about 200 pings no responses are received and the node in the middle will give a "error: packet buffer full". In my case the node in the middle it an 6LBR and the pinging node is Linux.

authmillenon, 2015-09-22:
Will try on native.

OlegHahm, 2015-09-22:
If I disconnect the destination node and print the address of `_first_unused` in the beginning of every IPv6 `send()` call, it will continuously increase for each call after the destination got disconnected. Aparently something is not freed correctly in the error case.

authmillenon, 2015-09-23:
Issue seems to be solved (right @OlegHahm?)

OlegHahm, 2015-09-23:
Partly.

OlegHahm, 2015-09-23:
The memory leak is fixed and I can still ping via the 6LBR. However, if I disconnect the 6LR and connect it again, it will stay in the neighbor cache of the 6LBR as `UNREACHABLE`. As I wrote, I can still ping the 6LR from Linux over the 6LBR, but the NCE says `UNREACHABLE`. As soon as I try to ping directly from 6LBR it changes to `REACHABLE` again...

authmillenon, 2015-09-23:
Maybe, because the 6LBR is using the 6LRs link-local address for forwarding?

OlegHahm, 2015-09-23:
Nope.

OlegHahm, 2015-09-23:
Interestingly enough after some time the NCE will go `UNREACHABLE` even during a normal ping from Linux. Pinging stops working and not even pinging from 6LBR is possible now.

OlegHahm, 2015-09-23:
State before is `DELAY` and then `PROBE`

authmillenon, 2015-09-23:
Then the 6LBR is trying reachability detection but fails for some reason

authmillenon, 2015-09-23:
What should happen:
* 6LBR looses connection to 6LR because it was switched off => marks 6LR entry as UNREACHABLE
* 6LR is switched on again => sends out initial router solicitation to search for upstream routers
* 6LBR receives this router solicitation => marks 6LR entry as STALE
* 6LBR sends router advertisement to 6LR => marks 6LR entry as DELAY
* (6LR registers its address with an NS, but that shouldn't be changing states on 6LBR)
* DELAY timeout => marks 6LR as PROBE and sends out unicast NS to 6LR
* 6LR receives NS from 6LBR => answers with NA
* 6LBR receives NA => marks 6LR entry as REACHABLE

OlegHahm, 2015-09-23:
I could see the probing NS and the corresponding NA in the sniffer, but still the entry goes to UNREACHABLE. (I see three NS from 6LBR and two matching NA from 6LR for the address.)

authmillenon, 2015-09-23:
Wait… it *goes to* UNREACHABLE? I thought it stayed UNREACHABLE.

OlegHahm, 2015-09-23:
Sorry, this is a bit confusing, I admit. Currently I'm talking about the case, where I do not disconnect any node. But it will still enter `UNREACHABLE` state sometimes.

authmillenon, 2015-09-23:
Ah… did not know that this case existed :D

authmillenon, 2015-09-23:
How did #3945 work out?

OlegHahm, 2015-09-23:
I'm testing

OlegHahm, 2015-09-23:
Seems to be improved - though NCEs are still staying `UNREACHABLE` unless I ping them directly.

OlegHahm, 2015-09-23:
And there is still a memory leak somewhere.

OlegHahm, 2015-09-23:
I just figured out that the actual setting of `REACHABLE` seems to be done from the shell.

authmillenon, 2015-09-24:
> I just figured out that the actual setting of REACHABLE seems to be done from the shell.

huh that's new.

authmillenon, 2015-09-24:
ah … no, because there is a pong received :-)

OlegHahm, 2015-09-24:
yupp

OlegHahm, 2015-09-24:
Actually, I'm not so sure if the current behavior is not even RFC compliant.

authmillenon, 2015-09-24:
Might be. The problem is more: does the algorithm in https://github.com/RIOT-OS/RIOT/issues/3933#issuecomment-142644265 happen?

OlegHahm, 2015-09-24:
As far as I can tell, yes.

authmillenon, 2015-09-24:
Then it is.

authmillenon, 2015-09-24:
Wait... what about

> Seems to be improved - though NCEs are still staying UNREACHABLE unless I ping them directly.

?

OlegHahm, 2015-09-24:
Hm, the question is: why can I ping them in the first place if they are unreachable? That they don't change their state based on forwarded pings is okay and that they are updated based on successfully received pongs is okay, too.

authmillenon, 2015-09-24:
> Hm, the question is: why can I ping them in the first place if they are unreachable? That they don't change their state based on forwarded pings is okay and that they are updated based on successfully received pongs is okay, too.

But the forwarded pings should at least trigger some kind of probing or am I wrong?

authmillenon, 2015-09-24:
Or the other way around: booting the 6LR again should STALE the node, since there is a RS coming from it.

OlegHahm, 2015-09-24:
As far as I can tell, probing is only done if the state is `DELAY` not from `UNREACHABLE`. Don't know if this is RFC compliant.

OlegHahm, 2015-09-24:
> Or the other way around: booting the 6LR again should STALE the node, since there is a RS coming from it.

True. Will investigate tomorrow. Bed is calling now.

authmillenon, 2015-09-24:
So #3953 did work?

closed 2015-09-24 by OlegHahm

OlegHahm, 2015-09-24:
Awesome, on current master I cannot see any more wrong NCEs or memory leaks! Nice job! :balloon:

PR #3940 (2015-09-23): gnrc_slip: reorder received packet correctly
Should fix #3933 at least partly.
PR #3957 (2015-09-24): 6lowpan: do garbage collect unreachable NCEs
RFC6775 says "if NUD on the router determines that the host is UNREACHABLE (based on the logic in [RFC4861]), the NCE SHOULD NOT be deleted but rather retained until the Registration Lifetime expires." Since this is a "SHOULD NOT" and not a "MUST NOT", we can still remove these NCEs to circumvent NCEs going into UNREACHABLE and never leave this state again. Actually removing these entries from the cache seems also be the better choice for a memory constrained device.

Fixes #3933.