RIOT-OS/RIOT #3941: kinetis: timer or xtimer is broken

gebart, 2015-09-23:
I get the following output when running tests/xtimer_msg on mulle (k60):

```
2015-09-23 14:22:21,325 - INFO # now=4295:986718 -> every 2.0s: Hello World
2015-09-23 14:22:21,327 - INFO # now=4295:989789 -> every 5.0s: This is a Test
2015-09-23 14:22:21,327 - INFO # sec=1 min=0 hour=0
2015-09-23 14:22:22,300 - INFO # now=8591:962724 -> every 2.0s: Hello World
2015-09-23 14:22:22,304 - INFO # now=8591:965795 -> every 5.0s: This is a Test
2015-09-23 14:22:22,306 - INFO # sec=2 min=0 hour=0
2015-09-23 14:22:23,309 - INFO # now=12887:938730 -> every 2.0s: Hello World
2015-09-23 14:22:23,313 - INFO # now=12887:941895 -> every 5.0s: This is a Test
2015-09-23 14:22:23,315 - INFO # sec=3 min=0 hour=0
2015-09-23 14:22:24,318 - INFO # now=799:3500013029 -> every 2.0s: Hello World
2015-09-23 14:22:24,322 - INFO # now=799:3500016377 -> every 5.0s: This is a Test
2015-09-23 14:22:24,324 - INFO # sec=4 min=0 hour=0
2015-09-23 14:22:25,327 - INFO # now=5095:3499989572 -> every 2.0s: Hello World
2015-09-23 14:22:25,332 - INFO # now=5095:3499993006 -> every 5.0s: This is a Test
2015-09-23 14:22:25,333 - INFO # sec=5 min=0 hour=0
2015-09-23 14:22:26,336 - INFO # now=9391:3499966288 -> every 2.0s: Hello World
2015-09-23 14:22:26,340 - INFO # now=9391:3499969723 -> every 5.0s: This is a Test
2015-09-23 14:22:26,342 - INFO # sec=6 min=0 hour=0
2015-09-23 14:22:27,345 - INFO # now=13687:3499943005 -> every 2.0s: Hello World
2015-09-23 14:22:27,350 - INFO # now=13687:3499946536 -> every 5.0s: This is a Test
2015-09-23 14:22:27,351 - INFO # sec=7 min=0 hour=0
2015-09-23 14:22:28,355 - INFO # now=1599:2704050720 -> every 2.0s: Hello World
2015-09-23 14:22:28,359 - INFO # now=1599:2704054156 -> every 5.0s: This is a Test
2015-09-23 14:22:28,361 - INFO # sec=8 min=0 hour=0
2015-09-23 14:22:29,364 - INFO # now=5895:2704027438 -> every 2.0s: Hello World
2015-09-23 14:22:29,368 - INFO # now=5895:2704030872 -> every 5.0s: This is a Test
2015-09-23 14:22:29,370 - INFO # sec=9 min=0 hour=0
2015-09-23 14:22:30,373 - INFO # now=10191:2704004154 -> every 2.0s: Hello World
2015-09-23 14:22:30,378 - INFO # now=10191:2704007676 -> every 5.0s: This is a Test
2015-09-23 14:22:30,380 - INFO # sec=10 min=0 hour=0
2015-09-23 14:22:31,383 - INFO # now=14487:2703981132 -> every 2.0s: Hello World
2015-09-23 14:22:31,387 - INFO # now=14487:2703984661 -> every 5.0s: This is a Test
2015-09-23 14:22:31,389 - INFO # sec=11 min=0 hour=0
2015-09-23 14:22:32,392 - INFO # now=2399:1908088932 -> every 2.0s: Hello World
2015-09-23 14:22:32,397 - INFO # now=2399:1908092367 -> every 5.0s: This is a Test
2015-09-23 14:22:32,399 - INFO # sec=12 min=0 hour=0
2015-09-23 14:22:33,402 - INFO # now=6695:1908065736 -> every 2.0s: Hello World
2015-09-23 14:22:33,406 - INFO # now=6695:1908069170 -> every 5.0s: This is a Test
2015-09-23 14:22:33,408 - INFO # sec=13 min=0 hour=0
2015-09-23 14:22:34,411 - INFO # now=10991:1908042539 -> every 2.0s: Hello World
2015-09-23 14:22:34,416 - INFO # now=10991:1908046069 -> every 5.0s: This is a Test
2015-09-23 14:22:34,418 - INFO # sec=14 min=0 hour=0
2015-09-23 14:22:35,421 - INFO # now=15287:1908019524 -> every 2.0s: Hello World
2015-09-23 14:22:35,425 - INFO # now=15287:1908023047 -> every 5.0s: This is a Test
2015-09-23 14:22:35,427 - INFO # sec=15 min=0 hour=0
```

so the 1 second timer works properly, but the other timers fire immediately with garbage values for `now`

I also got this below output on native _once_, when I ran the same `make term` again (nothing recompiled) it behaved normally again:

```
IOT native interrupts/signals initialized.
LED_GREEN_OFF
LED_RED_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2014.12-3571-gd9c44-mazarin-pr/cortexm-state-rebased)
This is thread 3
sending 1st msg
now=4294:967950 -> every 2.0s: Hello World
sending 2nd msg
now=4294:967968 -> every 5.0s: This is a Test
This is thread 4
sec=1 min=0 hour=0
now=4296:967992 -> every 2.0s: Hello World
sec=2 min=0 hour=0
sec=3 min=0 hour=0
now=4298:968042 -> every 2.0s: Hello World
now=8593:935563 -> every 5.0s: This is a Test
now=8593:935569 -> every 2.0s: Hello World
sec=4 min=0 hour=0
sec=5 min=0 hour=0
now=8595:935607 -> every 2.0s: Hello World
sec=6 min=0 hour=0
sec=7 min=0 hour=0
now=8597:935665 -> every 2.0s: Hello World
sec=8 min=0 hour=0
now=8598:935603 -> every 5.0s: This is a Test
sec=9 min=0 hour=0
now=8599:935773 -> every 2.0s: Hello World
sec=10 min=0 hour=0
sec=11 min=0 hour=0
now=8601:935832 -> every 2.0s: Hello World
sec=12 min=0 hour=0
sec=13 min=0 hour=0
now=8603:935662 -> every 5.0s: This is a Test
now=8603:935888 -> every 2.0s: Hello World
sec=14 min=0 hour=0
sec=15 min=0 hour=0
now=8605:935957 -> every 2.0s: Hello World
sec=16 min=0 hour=0
sec=17 min=0 hour=0
now=8607:936015 -> every 2.0s: Hello World
sec=18 min=0 hour=0
now=8608:935767 -> every 5.0s: This is a Test
sec=19 min=0 hour=0
now=8609:936062 -> every 2.0s: Hello World
sec=20 min=0 hour=0
sec=21 min=0 hour=0
now=8611:936142 -> every 2.0s: Hello World
sec=22 min=0 hour=0
sec=23 min=0 hour=0
now=8613:935817 -> every 5.0s: This is a Test
now=8613:936161 -> every 2.0s: Hello World
sec=24 min=0 hour=0
sec=25 min=0 hour=0
now=8615:936259 -> every 2.0s: Hello World
sec=26 min=0 hour=0
sec=27 min=0 hour=0
now=8617:936311 -> every 2.0s: Hello World
sec=28 min=0 hour=0
now=8618:935924 -> every 5.0s: This is a Test
sec=29 min=0 hour=0
now=8619:936420 -> every 2.0s: Hello World
sec=30 min=0 hour=0
sec=31 min=0 hour=0
now=8621:936499 -> every 2.0s: Hello World
sec=32 min=0 hour=0
sec=33 min=0 hour=0
now=8623:936033 -> every 5.0s: This is a Test
now=8623:936558 -> every 2.0s: Hello World
sec=34 min=0 hour=0
sec=35 min=0 hour=0
now=8625:936632 -> every 2.0s: Hello World
sec=36 min=0 hour=0
sec=37 min=0 hour=0
now=8627:936686 -> every 2.0s: Hello World
sec=38 min=0 hour=0
now=8628:936143 -> every 5.0s: This is a Test
sec=39 min=0 hour=0
now=8629:936794 -> every 2.0s: Hello World
sec=40 min=0 hour=0
sec=41 min=0 hour=0
now=8631:936902 -> every 2.0s: Hello World
sec=42 min=0 hour=0
sec=43 min=0 hour=0
now=8633:936206 -> every 5.0s: This is a Test
now=8633:936946 -> every 2.0s: Hello World
sec=44 min=0 hour=0
``` 
I don't know if there is a race inside the xtimer code which happened to appear once on native or if the problem lies inside periph_timer on k60.

It seems like this bug causes a flood of network packets when running RPL in gnrc_networking on mulle because of timeouts and retransmissions and trickle timer timeouts

kaspar030, 2015-09-23:
@gebart Could you try printing xtimer_now() in a loop and check if the values make sense?

gebart, 2015-09-23:
@kaspar030 I did two tests, xtimer_now seem to make sense.

First test:

```c
    while(1) {
        xtimer_usleep(1000000ul);
        printf("now: %" PRIu32 "\n", xtimer_now());
    }
```

```
2015-09-23 19:35:39,290 - INFO # This is thread 3
2015-09-23 19:35:40,285 - INFO # now: 1003244
2015-09-23 19:35:41,280 - INFO # now: 1998663
2015-09-23 19:35:42,276 - INFO # now: 2994079
2015-09-23 19:35:43,271 - INFO # now: 3989494
2015-09-23 19:35:44,266 - INFO # now: 4984909
2015-09-23 19:35:45,261 - INFO # now: 5980325
2015-09-23 19:35:46,257 - INFO # now: 6975739
2015-09-23 19:35:47,252 - INFO # now: 7971155
2015-09-23 19:35:48,247 - INFO # now: 8966570
2015-09-23 19:35:49,243 - INFO # now: 9961984
2015-09-23 19:35:50,238 - INFO # now: 10957400
2015-09-23 19:35:51,234 - INFO # now: 11952910
2015-09-23 19:35:52,229 - INFO # now: 12948413
2015-09-23 19:35:53,224 - INFO # now: 13943924
```

and a tight loop:
```c
    while(1) {
        printf("now: %" PRIu32 "\n", xtimer_now());
    }
```

```
2015-09-23 19:36:11,059 - INFO # main(): This is RIOT! (Version: 2014.12-3571-gd9c44-mazarin-pr/cortexm-state-rebased)
2015-09-23 19:36:11,061 - INFO # This is thread 3
2015-09-23 19:36:11,061 - INFO # now: 8229
2015-09-23 19:36:11,062 - INFO # now: 9091
2015-09-23 19:36:11,063 - INFO # now: 9959
2015-09-23 19:36:11,064 - INFO # now: 10827
2015-09-23 19:36:11,065 - INFO # now: 11781
2015-09-23 19:36:11,066 - INFO # now: 12736
2015-09-23 19:36:11,067 - INFO # now: 13690
2015-09-23 19:36:11,068 - INFO # now: 14645
2015-09-23 19:36:11,069 - INFO # now: 15599
2015-09-23 19:36:11,070 - INFO # now: 16553
2015-09-23 19:36:11,071 - INFO # now: 17508
2015-09-23 19:36:11,072 - INFO # now: 18462
2015-09-23 19:36:11,072 - INFO # now: 19417
2015-09-23 19:36:11,073 - INFO # now: 20371
2015-09-23 19:36:11,075 - INFO # now: 21326
2015-09-23 19:36:11,075 - INFO # now: 22280
2015-09-23 19:36:11,077 - INFO # now: 23235
2015-09-23 19:36:11,078 - INFO # now: 24189
2015-09-23 19:36:11,078 - INFO # now: 25144
2015-09-23 19:36:11,079 - INFO # now: 26098
2015-09-23 19:36:11,080 - INFO # now: 27053
2015-09-23 19:36:11,081 - INFO # now: 28007
2015-09-23 19:36:11,082 - INFO # now: 28962
2015-09-23 19:36:11,083 - INFO # now: 29916
2015-09-23 19:36:11,084 - INFO # now: 30871
2015-09-23 19:36:11,085 - INFO # now: 31825
2015-09-23 19:36:11,086 - INFO # now: 32780
2015-09-23 19:36:11,087 - INFO # now: 33734
2015-09-23 19:36:11,088 - INFO # now: 34689
2015-09-23 19:36:11,089 - INFO # now: 35643
2015-09-23 19:36:11,090 - INFO # now: 36598
2015-09-23 19:36:11,091 - INFO # now: 37552
2015-09-23 19:36:11,092 - INFO # now: 38507
2015-09-23 19:36:11,093 - INFO # now: 39461
2015-09-23 19:36:11,093 - INFO # now: 40415
2015-09-23 19:36:11,094 - INFO # now: 41370
2015-09-23 19:36:11,095 - INFO # now: 42324
```

gebart, 2015-09-23:
The usleep seem to end a bit early though, is that common?

gebart, 2015-09-23:
This is a heisenbug...
When I turn on ENABLE_DEBUG inside xtimer-core.c it works properly (because of the added delays from the printfs I guess).

```
2015-09-23 19:17:28,554 - INFO # This is thread 3
2015-09-23 19:17:28,555 - INFO # sending 1st msg
2015-09-23 19:17:28,558 - INFO # now=0:12760 -> every 2.0s: Hello World
2015-09-23 19:17:28,562 - INFO # timer_set(): offset=2000000 now=16139 (16139)
2015-09-23 19:17:28,566 - INFO # timer_set_absolute(): now=20131 target=2020130
2015-09-23 19:17:28,572 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:28,577 - INFO # timer_set_absolute(): timer is new list head. updating lltimer.
2015-09-23 19:17:28,580 - INFO # __lltimer_set(): setting 2020110
2015-09-23 19:17:28,582 - INFO # sending 2nd msg
2015-09-23 19:17:28,585 - INFO # now=0:39399 -> every 5.0s: This is a Test
2015-09-23 19:17:28,589 - INFO # timer_set(): offset=5000000 now=43038 (43038)
2015-09-23 19:17:28,593 - INFO # timer_set_absolute(): now=47030 target=5047029
2015-09-23 19:17:28,599 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:28,600 - INFO # This is thread 4
2015-09-23 19:17:28,604 - INFO #  _xtimer_set64() offset=1000000 long_offset=0
2015-09-23 19:17:28,608 - INFO # timer_set(): offset=1000000 now=61953 (61954)
2015-09-23 19:17:28,612 - INFO # timer_set_absolute(): now=65946 target=1065945
2015-09-23 19:17:28,618 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:28,623 - INFO # timer_set_absolute(): timer is new list head. updating lltimer.
2015-09-23 19:17:28,626 - INFO # __lltimer_set(): setting 1065925
2015-09-23 19:17:29,612 - INFO # _timer_callback() now=1065926 (1065926)pleft=4293901368
2015-09-23 19:17:29,615 - INFO # __lltimer_set(): setting 2020110
2015-09-23 19:17:29,617 - INFO # sec=1 min=0 hour=0
2015-09-23 19:17:29,621 - INFO #  _xtimer_set64() offset=1000000 long_offset=0
2015-09-23 19:17:29,625 - INFO # timer_set(): offset=1000000 now=1078663 (1078663)
2015-09-23 19:17:29,629 - INFO # timer_set_absolute(): now=1083002 target=2083001
2015-09-23 19:17:29,635 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:30,566 - INFO # _timer_callback() now=2020111 (2020111)pleft=4292947183
2015-09-23 19:17:30,569 - INFO # __lltimer_set(): setting 2082981
2015-09-23 19:17:30,573 - INFO # now=2:27211 -> every 2.0s: Hello World
2015-09-23 19:17:30,577 - INFO # timer_set(): offset=2000000 now=2030590 (2030590)
2015-09-23 19:17:30,581 - INFO # timer_set_absolute(): now=2034929 target=4034928
2015-09-23 19:17:30,587 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:30,629 - INFO # _timer_callback() now=2082982 (2082982)pleft=4292884312
2015-09-23 19:17:30,632 - INFO # __lltimer_set(): setting 4034908
2015-09-23 19:17:30,634 - INFO # sec=2 min=0 hour=0
2015-09-23 19:17:30,638 - INFO #  _xtimer_set64() offset=1000000 long_offset=0
2015-09-23 19:17:30,642 - INFO # timer_set(): offset=1000000 now=2095719 (2095719)
2015-09-23 19:17:30,646 - INFO # timer_set_absolute(): now=2100058 target=3100058
2015-09-23 19:17:30,652 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:30,657 - INFO # timer_set_absolute(): timer is new list head. updating lltimer.
2015-09-23 19:17:30,660 - INFO # __lltimer_set(): setting 3100038
2015-09-23 19:17:31,646 - INFO # _timer_callback() now=3100039 (3100039)pleft=4291867255
2015-09-23 19:17:31,649 - INFO # __lltimer_set(): setting 4034908
2015-09-23 19:17:31,651 - INFO # sec=3 min=0 hour=0
2015-09-23 19:17:31,655 - INFO #  _xtimer_set64() offset=1000000 long_offset=0
2015-09-23 19:17:31,659 - INFO # timer_set(): offset=1000000 now=3112776 (3112777)
2015-09-23 19:17:31,663 - INFO # timer_set_absolute(): now=3117116 target=4117115
2015-09-23 19:17:31,669 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:32,581 - INFO # _timer_callback() now=4034909 (4034909)pleft=4290932385
2015-09-23 19:17:32,584 - INFO # __lltimer_set(): setting 4117095
2015-09-23 19:17:32,587 - INFO # now=4:42009 -> every 2.0s: Hello World
2015-09-23 19:17:32,592 - INFO # timer_set(): offset=2000000 now=4045388 (4045389)
2015-09-23 19:17:32,596 - INFO # timer_set_absolute(): now=4049727 target=6049727
2015-09-23 19:17:32,601 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:32,663 - INFO # _timer_callback() now=4117096 (4117096)pleft=4290850198
2015-09-23 19:17:32,666 - INFO # __lltimer_set(): setting 5047009
2015-09-23 19:17:32,668 - INFO # sec=4 min=0 hour=0
2015-09-23 19:17:32,672 - INFO #  _xtimer_set64() offset=1000000 long_offset=0
2015-09-23 19:17:32,676 - INFO # timer_set(): offset=1000000 now=4129831 (4129832)
2015-09-23 19:17:32,680 - INFO # timer_set_absolute(): now=4134171 target=5134170
2015-09-23 19:17:32,686 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:33,593 - INFO # _timer_callback() now=5047010 (5047010)pleft=4289920284
2015-09-23 19:17:33,596 - INFO # __lltimer_set(): setting 5134150
2015-09-23 19:17:33,600 - INFO # now=5:54110 -> every 5.0s: This is a Test
2015-09-23 19:17:33,604 - INFO # timer_set(): offset=5000000 now=5057749 (5057750)
2015-09-23 19:17:33,608 - INFO # timer_set_absolute(): now=5062088 target=10062088
2015-09-23 19:17:33,614 - INFO # timer_set_absolute(): timer will expire in this timer period.
[... cut ...]
2015-09-23 19:17:34,706 - INFO #  _xtimer_set64() offset=1000000 long_offset=0
2015-09-23 19:17:34,710 - INFO # timer_set(): offset=1000000 now=6163943 (6163943)
2015-09-23 19:17:34,714 - INFO # timer_set_absolute(): now=6168282 target=7168282
2015-09-23 19:17:34,720 - INFO # timer_set_absolute(): timer will expire in this timer period.
2015-09-23 19:17:34,725 - INFO # timer_set_absolute(): timer is new list head. updating lltimer.
2015-09-23 19:17:34,728 - INFO # __lltimer_set(): setting 7168262
```

gebart, 2015-09-24:
bisection gave:

```
a719e7d61ff90ce3c40a6508a5ed5af6b873cfe9 is the first bad commit
commit a719e7d61ff90ce3c40a6508a5ed5af6b873cfe9
Author: Kaspar Schleiser <kaspar@schleiser.de>
Date:   Tue Sep 15 00:11:28 2015 +0200

    sys: xtimer: some updates
    
    - more robust underflow protection in xtimer_usleep_until()
    - use relative target in xtimer_spin()
    - honour reference in isr when spinning until timer target
    - add XTIMER_BACKOFF to xtimer_spin_until() target when backing off in
      _timer_set_absolute()
    - doxygen updates

:040000 040000 0a65e396706db616c1b088af4e4c86c790c37b0d 6c6a912b2188500a70324117d44d6b0d4baadc72 M      sys
```

gebart, 2015-09-24:
Further manual testing pinpoints the fault to line 455 of xtimer_core.c
https://github.com/RIOT-OS/RIOT/commit/a719e7d61ff90ce3c40a6508a5ed5af6b873cfe9#diff-5371c49e6244a42b30cd3a714c0e9aacR455

Reverting this part of the patch makes the xtimer_msg test work properly on Kinetis:

```diff
--- b/sys/xtimer/xtimer_core.c
+++ a/sys/xtimer/xtimer_core.c
@@ -452,7 +452,7 @@ overflow:
     /* check if next timers are close to expiring */
     while (timer_list_head && (_time_left(_mask(timer_list_head->target), reference) < XTIMER_ISR_BACKOFF)) {
         /* make sure we don't fire too early */
-        while (_time_left(_mask(timer_list_head->target), 0));
+        while (_time_left(_mask(timer_list_head->target), reference));
 
         /* pick first timer in list */
         xtimer_t *timer = timer_list_head;
```

I have not dug any deeper yet, maybe @kaspar030 has an idea what may be wrong given this information.

kaspar030, 2015-09-24:
> The usleep seem to end a bit early though, is that common?

Actually, no, it shouldn't end early.

kaspar030, 2015-09-24:
> Further manual testing pinpoints the fault to line 455 of xtimer_core.c

Thanks a lot for digging!
That line *shouldn't* cause any problem, as the same logic is used in the while() above.

I still suspect that something with the timer read is wrong.
Could you try the tight-loop test again, but with a second thread doing very short (10000 to 1000 usec) xtimer-usleeps in a loop? I wish I had a mulle board...

kaspar030, 2015-09-24:
I have a similar platform with a similar (downcounting) timer scheme (itron M3). I ended up using one channel solely for reading (just returning "0-VAL" to get an upcounting value), and the second channel to create the interrupts, combining both to one logical RIOT timer.

gebart, 2015-09-24:
I think the issue lies in the periph/timer implementation and is something similar to the problem experienced by the hwtimer in #3139 and were mitigated by #3146

A similar solution may fix the xtimer problems.

gebart, 2015-09-24:
I don't want this problem to hold the release back unnecessarily with this problem since a proper solution is in the works with using LPTMR instead to allow counting time in low-power modes, but it will be too large a change to introduce this late in the release cycle.

@kaspar030 does your solution for the itron with two periph/timer devices need any modifications to xtimer or simply configuration?

haukepetersen, 2015-09-24:
I think I might now the reason, looks pretty similar with the stuff I was fighting with for the `nucleo-l1` board: It seems, that the default xtimer configuration is not valid for the `mulle` board. 

@gebart, could you try and add
```c

/**
 * @name xtimer configuration
 * @{
 */
#define XTIMER_OVERHEAD     (6)
#define XTIMER_BACKOFF      (3)
/** @} */
```c
to the `mulle/board.h`? I have no idea if these are actually the perfect values, but they seem to work on all boards where I had trouble with the xtimer. @kaspar030 can explain them probably better?!

In general: I think it make sense to make the xtimer configuration explicit to the boards, meaning that every board has to define these values instead of having some default values that seem to create problems an many boards...

kaspar030, 2015-09-24:
> does your solution for the itron with two periph/timer devices need any modifications to xtimer or is it simply configuration?

No, it doesn't need xtimer changes.

The itron timers seem to be very similar to the kinetis ones, they can downcount from a specific value, generating an interrupt when they arrive at zero and resetting to their initially set value. Also, the semantics when re-setting timers seem similar (like, writing a new start value will be effective after the the timer re-reads it when reaching zero, so timers need to be deactivated first on re-set).

I just use the first timer set to down-count from 0xffffffff, with an empty ISR. When any of the timers want to read out the current value, I return the first timers value (actually, 0-value to make it upcounting).

Now with the other timer channels, when setting a relative value, I set that value and let it count to 0. for absolute sets, I set it to "absolute_value-first_timer_now".

That way, I lose one timer channel, but the code doesn't have to do any complex arithmetic and doesn't ever loose ticks.

This is all at periph/timer level, so that interfaces exposes n-1 timers.

gebart, 2015-09-25:
I have prepared a change PR, but I am currently on the fly and without hardware to test my implementation on.

gebart, 2015-09-25:
I think the problem is with the `_time_left` function in xtimer-core.c. it will return zero if `now < reference`, which it will be since the hardware timer is set to fire early and we have not yet spun to the right time.

kaspar030, 2015-09-25:
> I think the problem is with the |_time_left| function in xtimer-core.c.
> it will return zero if |now < reference|, which it will be since the
> hardware timer is set to fire early and we have not yet spun to the
> right time.

That is correct. I'll need to revert that change.

PR #3971 (2015-09-26): xtimer-core: Use xtimer_now as reference time instead of future timer target
Fixes #3941 for me.

I believe this should fix the problem with firing remaining timers immediately when the first timer fires.

The function `_timer_callback` in xtimer-core.c will be called before the actual time (depending on the setting `XTIMER_OVERHEAD`). If we set the reference to the timer target instead of `xtimer_now` the `_time_left` function believes that we have wrapped around to the next long period, resulting in firing of all timers queued for the current period as well as a long term tick increment.
closed 2015-09-26 by gebart