RIOT-OS/RIOT #3943: samr21-xpro: UART hangs

OlegHahm, 2015-09-23:
On samr21-xpro using the `gnrc_border_router` example, it happens that the shell will stop working. Connecting with gdb indicates that it hangs in
```
uart_write_blocking (uart=uart@entry=UART_0, data=62 '>') at /home/oleg/git/RIOT/cpu/samd21/periph/uart.c:135
135	    while (!(_uart(uart)->INTFLAG.reg & SERCOM_USART_INTFLAG_DRE));
```
while a backtrace shows that this originates from
```
#0  uart_write_blocking (uart=uart@entry=UART_0, data=62 '>') at /home/oleg/git/RIOT/cpu/samd21/periph/uart.c:135
#1  0x00000604 in uart_stdio_write (
    buffer=0x200060d9 " ba::1", ' ' <repeats 26 times>, "5  36:32:48:33:46:dd:7c:2a   UNREACHABLE REG\n\nSMA Retries: 4 \n\255\347\n:2G\203\254\373\325\335\305k\371\311\345\n\233\351\210S\333B\201>\361r\274Zx\252\357\341\t\256\312\061\f}\366}\366\367\370خ\v'\247\257_\273<\305{\343\303\337Ⱥ\327\335ǊŘ\207#\271\005-/)H;۷\311\367O\337\036\301^\304\363T\243\365\363\205\352\222\067\346L\315\315\342y\216\200_\362\036`"..., len=2) at /home/oleg/git/RIOT/sys/uart_stdio/uart_stdio.c:94
#2  0x0000010a in _write_r (r=<optimized out>, fd=<optimized out>, data=<optimized out>, count=<optimized out>) at /home/oleg/git/RIOT/sys/newlib/syscalls.c:204
#3  0x0000cf86 in __swrite ()
#4  0x0000c93e in __sflush_r ()
#5  0x0000c99e in _fflush_r ()
#6  0x0000c9c6 in fflush ()
#7  0x00007aa8 in print_prompt () at /home/oleg/git/RIOT/sys/shell/shell.c:271
#8  0x00007ac0 in shell_run (shell_commands=0xf27 <main+26>, shell_commands@entry=0xe288 <shell_commands>, line_buf=0x61006568 "", line_buf@entry=0x20000800 <main_stack+1332> "ncache", 
    len=5, len@entry=128) at /home/oleg/git/RIOT/sys/shell/shell.c:275
#9  0x00000f26 in main () at /home/oleg/git/RIOT/examples/gnrc_border_router/main.c:46
```

OlegHahm, 2015-09-23:
Introducing a safeguard counter variable in the while loop helps to work around the problem.

daniel-k, 2015-09-24:
Is it outputting `>` all the time? I had this problem once before but was using my self-ported non-blocking uart. I blamed newlib back then after some debugging. Never really solved it.

OlegHahm, 2015-09-24:
Sometimes, but most often it just hangs.

kaspar030, 2015-09-24:
Could you check if it is entering and exiting uart_stdio_write, e.g., if shell is trying to print '>' in a tight loop?
that would mean the reading side is broken.

daniel-k, 2015-09-24:
@kaspar030 for me it was reproducible with pressing up/down arrow keys. But that doesn't happen anymore on `master`. That was actually why I blamed newlib. Maybe this information helps.

OlegHahm, 2015-09-24:
@daniel-k, which terminal program were you using?

daniel-k, 2015-09-24:
I use picocom:
`picocom --imap lfcrlf -b 115200 /dev/...`

OlegHahm, 2015-09-24:
> Could you check if it is entering and exiting uart_stdio_write, e.g., if shell is trying to print '>' in a tight loop?

Confirmed.

OlegHahm, 2015-09-24:
On reliable way of reproducing it: build `gnrc_border_router` example, flash it, call make term, enter `ncache` and keep the enter button pressed (causing pyterm to send the line over and over again).

kaspar030, 2015-09-24:
Tried it, still working fine after 10 seconds...

OlegHahm, 2015-09-24:
Yupp, just figured out, there need to be traffic on the second UART, I guess.

OlegHahm, 2015-09-24:
Or maybe not even that. Pinging from another node seems also cause the trouble.

OlegHahm, 2015-09-24:
> Is it outputting > all the time? I had this problem once before but was using my self-ported non-blocking uart. I blamed newlib back then after some debugging. Never really solved it.

Sorry, my fault, yet it prints `>` all the time, just pyterm is only outputting stuff after a `\n`

kaspar030, 2015-09-24:
@OlegHahm Could you try #3923?

daniel-k, 2015-09-24:
When I had this I tracked it down to `readline` returning immediately. Maybe something down `getchar` doesn't like to be interrupted?

kaspar030, 2015-09-24:
This probably means that the samd21's uart receiveing implementation doesn't clear the interrupt flags correctly.

OlegHahm, 2015-09-24:
Tried . Seems to help.

PeterKietzmann, 2015-09-25:
#3923 is merged can we close that issue

OlegHahm, 2015-09-25:
I guess so, yes. Haven't seen the problem with #3923. However, it might still point to another problem in the UART implementation of the SAM.

kaspar030, 2015-12-18:
Did this show up again?

OlegHahm, 2015-12-18:
Nope. However, to me it seems that the underlying problem might still not be fixed. But I would be okay with closing this one.

haukepetersen, 2015-12-18:
Then let's close this for know. (Also the UART driver was updated since then, so that might have resolved the issue?)

OlegHahm, 2015-12-18:
Why is everybody talking about closing it but no-one does it? ;-)

closed 2015-12-18 by OlegHahm