RIOT-OS/RIOT #3947: doc: periph_timer redefine tim_t

A-Paul, 2015-09-23:
In https://github.com/RIOT-OS/RIOT/pull/3780 it turned out that `tim_t` could not be used in a general test, because this type may be redefined in a cpu-specific implementation.

It should be documented how this replacement would be implemented correctly, and what has to be taken care off, to not break existing code.

miri64, 2016-10-17:
@haukepetersen bump?

OlegHahm, 2017-01-15:
Actually, I don't understand the problem.

PeterKietzmann, 2017-01-15:
I think @A-Paul refers to [this comment](https://github.com/RIOT-OS/RIOT/pull/3780#r39371758)  and asks for enhancement of the documentation [here](https://github.com/RIOT-OS/RIOT/blob/8a74475e70925db155ee9747d514004f8d2dadd7/drivers/include/periph/timer.h#L37)?

A-Paul, 2017-01-17:
@PeterKietzmann, your assumption is correct. :-)
To be more precise,I followed this advice: https://github.com/RIOT-OS/RIOT/pull/3780#r39622407.

PeterKietzmann, 2017-01-18:
And did you see your issue solved somewhere during all the documentation efforts? I didn't look it up...

kYc0o, 2018-01-29:
A gentle ping for those who are involved here. I saw some PRs regarding this issue, however I don't know if we can mark it as solved.

Can someone confirm the doc still missing?

haukepetersen, 2018-01-30:
No, I think this issue is still valid to some point. Though I think this is not so much about the specific `tim_t` type, but what is (still) missing is a more coherent documentation about the general peripheral driver concept, specifically on the concept of platform specific types this concept is using. 

So anyone volunteering to write something up? I won't get to this anytime soon...

closed 2019-09-10 by stale[bot]