RIOT-OS/RIOT #3967: periph/timer: Drop timer_reset() ?

gebart, 2015-09-25:
I propose to remove this function from the periph/timer API.

```c
/**
 * @brief Reset the up-counting value to zero for the given timer
 *
 * Note that this function effects all currently set channels and it can lead to
 * non-deterministic timeouts if any channel is active when this function is
 * called.
 *
 * @param[in] dev           the timer to reset
 */
void timer_reset(tim_t dev);
```

It is currently implemented for all periph/timer implementations, but not used anywhere. The rationale behind removing it is that the warning in the API documentation makes this function unsuitable for any kind of reusable code in my opinion. The problem it solves can just as easily be solved by using the xtimer API or using timer stop/start in sequence.

gebart, 2015-10-14:
ping @kaspar030

kaspar030, 2015-10-14:
+1, let's get rid of it.

gebart, 2015-10-17:
Implemented in #4104.
the OpenWSN patches seem to have some reference to timer_reset (and also hwtimer). I remember seeing @thomaseichinger or @OlegHahm referring to an update on the OpenWSN support in RIOT, was that update merged already?

PeterKietzmann, 2015-10-26:
#4104 was merged

closed 2015-10-26 by PeterKietzmann