RIOT-OS/RIOT #3970: RPL: Advertise DODAG only over the assigned interface

authmillenon, 2015-09-25:
`gnrc_rpl` seems to multicast DIOs over all interfaces, though `gnrc_rpl_init` expects an interface as parameter and sets the RPL-nodes multicast address only for that interface. An easy fix would be to set the interface in the `gnrc_rpl_dodag_t` struct and use it in `gnrc_rpl_send()` to generate a `gnrc_netif_hdr` that addresses that interface.

OlegHahm, 2015-09-25:
Off-topic: what's the difference between the `network` and `NSTF` label these days?

authmillenon, 2015-09-25:
I generally set `NSTF` for all gnrc related stuff (we might need to rename it), and `network` for all network stuff in general

A-Paul, 2015-12-16:
Hi @authmillenon, is this fixed by #3972 or is it still an issue.

authmillenon, 2015-12-16:
AFAIK no. #3972 did not fix this issue, it just made usage of multiple interfaces possible without crashing due to this bug.

OlegHahm, 2015-12-18:
@cgundogan, is this something that should into the release notes as a known issue?

closed 2016-03-23 by cgundogan

cgundogan, 2016-03-23:
@authmillenon this should be fixed now, can you verify?

PR #5135 (2016-03-23): gnrc_rpl: proper interface handling
At DODAG creation, the appropriate interface id will be assigned to the DODAG.
This way, it is not necessary to look up the correct interface id at several places. Furthermore, looking up the interface id like it was done before had some flaws when multiple interfaces were present.

fixes #3970