RIOT-OS/RIOT #4025: core: preserve disabled IRQs over context switches

kaspar030, 2015-10-02:
Does anyone have a use case for preserving "globally disabled IRQ" state over context switches?

Currently, some platforms do it, some don't, and AFAIK, this "feature" is needed nowhere. If we would hard enable IRQ's on any user-invoked context switch, we could simplify a lot of the IPC code.

haukepetersen, 2016-02-09:
Hm, seems to me that it is not very useful. If I disable IRQs somewhere in the code, I do this to prevent context switches, right?

Also on Cortex-M platforms, context switches NEED global IRQs to be enabled to work, otherwise the PendSV interrupt (which is doing the context switch) would never be triggered... So this 'feature' is not possible on these platforms anyhow as I see it

kaspar030, 2016-02-09:
@haukepetersen Exactly. Current code could be simplified a lot if it could expect that a manually triggered context switch re-enables IRQs, that's why I'm asking.

haukepetersen, 2016-02-09:
So I don't see anything that speaks against this, and as it simplifies stuff, I am of course for it!

kaspar030, 2016-07-30:
marked as memo, closing for now.

closed 2016-07-30 by kaspar030