RIOT-OS/RIOT #4026: How to debug crash/freeze on the MSBA2?

x3ro, 2015-10-02:
I'm trying to implement file system support for the MSBA2 (being the only board at my disposal with an SD card slot an an implemented driver for accessing it) and am running into a crash that I don't know how to debug:

    static struct file * find_file(const char *name)
    {
      int i;
      struct file_header hdr;
      coffee_page_t page;

      [...]

      printf("find file loop start\n");

      /* Scan the flash memory sequentially otherwise. */
      for(page = 0; page < COFFEE_PAGE_COUNT; page = next_file(page, &hdr)) {
        read_header(&hdr, page);
        if(HDR_ACTIVE(hdr) && !HDR_LOG(hdr) && strcmp(name, hdr.name) == 0) {
          return load_file(page, &hdr);
        }
      }
      printf("find file loop end\n");

      // Crash happens happens somewhere here

      return NULL;
    }

If the content of the loop is removed, it doesn't crash. If it is active, it simply gets stuck after the `printf`, i.e. the place the `return` should return to is never reached. I honestly have no idea how to debug this since the MSBA2 does not seem to have debugger support :(

miri64, 2016-10-18:
Is this still an issue over a year later?

x3ro, 2016-10-18:
I suppose the concrete issue is fixed, but this was more of an open-ended question regarding how to properly debug on a platform w/o gdb support. But I suppose it can be closed.

closed 2016-10-18 by x3ro

miri64, 2016-10-18:
`DEBUG()` macro and [shotgun debugging](https://en.wikipedia.org/wiki/Shotgun_debugging) is usually my preferred method for such cases. ;-)