RIOT-OS/RIOT #4027: Hitting assert when disabling `6lo` on interface (iotlab-m3)

cgundogan, 2015-10-02:
When issuing `ifconfig 7 -6lo` on an iotlab-m3 hits the assert in [gnrc_sixlowpan_nd.c#L66](https://github.com/RIOT-OS/RIOT/blob/ae46d5d6a030ff1237d0263da68999f566fb2015/sys/net/gnrc/network_layer/sixlowpan/nd/gnrc_sixlowpan_nd.c#L66)

OlegHahm, 2015-10-02:
Seems sensible to me.

OlegHahm, 2015-10-02:
Configuring this at runtime seems rather pointless to me.

cgundogan, 2015-10-02:
can you elaborate? the `6lo` option should be removed for these cases then .. otherwise this is not user friendly (:

OlegHahm, 2015-10-02:
Yes, it should be removed.

authmillenon, 2015-10-02:
I also would vote to remove the `6lo` flag from `ifconfig`

authmillenon, 2015-10-02:
(it made sense when this wasn't configured automatically. Now it is, so just throw it out.)

authmillenon, 2015-10-02:
See #4028 in case this was missed.

PR #4028 (2015-10-02): shell: ifconfig: fixes
Two fixes to `ifconfig`:

* make 6lo flag read-only (fixes #4027)
* remove unused variable
closed 2015-10-02 by OlegHahm