RIOT-OS/RIOT #4037: shell: Add ISR stack support for ps shell command

gebart, 2015-10-03:
While the ISR is not a thread, it is still useful to be able to see stack usage for the interrupt context.

Example output from current system:
```
2015-10-03 10:17:58,836 - INFO # > ps
2015-10-03 10:17:58,843 - INFO #        pid | name                 | state    Q | pri | stack ( used) | location   
2015-10-03 10:17:58,849 - INFO #          1 | idle                 | pending  Q |  15 |   256 (  140) | 0x1fff0a48 
2015-10-03 10:17:58,856 - INFO #          2 | main                 | running  Q |   7 |  1536 (  792) | 0x1fff0b48 
2015-10-03 10:17:58,863 - INFO #          3 | pktdump              | bl rx    _ |   6 |  1536 (  244) | 0x1fff3d60 
2015-10-03 10:17:58,870 - INFO #          4 | 6lo                  | bl rx    _ |   3 |  1024 (  408) | 0x1fff4364 
2015-10-03 10:17:58,876 - INFO #          5 | ipv6                 | bl rx    _ |   4 |  1024 (  396) | 0x1fff16f8 
2015-10-03 10:17:58,883 - INFO #          6 | udp                  | bl rx    _ |   5 |  1024 (  272) | 0x1fff4b0c 
2015-10-03 10:17:58,890 - INFO #          7 | at86rfxx             | bl rx    _ |   3 |  1024 (  320) | 0x1fff12b8 
2015-10-03 10:17:58,895 - INFO #            | SUM                  |            |     |  7424 ( 2572)
```

Add something like 
```
x | interrupt | | -1 | 1234 (123) | 0xdeadbeef
```

MohmadAyman, 2015-10-08:
i tried working on this, i can now show if it is in an ISR or not, but i'm still having a hard time figuring how to get the stack size since it isn't a thread

gebart, 2015-10-08:
hmm, maybe this is not so straightforward as I first thought, it may take some remodeling in order to be able to make this platform independent...

For the Cortex-M platforms you need to look at the _sstack and _estack symbols to find the limits of the ISR stack. A similar algorithm as in the stack usage test for the threads can be used to find the usage of the stack. The stack is initially filled with a canary value, STACK_CANARY_WORD in vectors_cortexm.c.

MohmadAyman, 2015-10-09:
given your hints, and after looking at vectors_cortexm.c and the already implemented algorithm for calculating the thread stack, it wont take much to do it for the Cortex-M i guess, Do i open a PR for it ?
Or do i wait until i figure a way to make it platform independent first ?

gebart, 2015-10-24:
At the moment I don't really have a good platform independent approach. It might be possible to introduce this platform specifically using some define, e.g.
```
#if ARCH_HAS_ISR_STACK_USAGE
long isr_usage = isr_stack_usage();
printf("stack usage: %d\n", isr_usage);
#endif
```
and create the `isr_stack_usage` function inside cpu/cortexm_common

MohmadAyman, 2015-10-24:
Okay,  gonna start working on it this weekend.

haukepetersen, 2015-12-09:
for reference: #4201

haukepetersen, 2016-04-20:
postponed

kYc0o, 2016-07-26:
Solved by #5491

closed 2016-07-26 by kYc0o