RIOT-OS/RIOT #4048: gnrc: potential racey memory leak

OlegHahm, 2015-10-05:
According to the `gnrc_pktbuf_stats()` output the packet buffer contains some chunk when it should be empty. This behavior can be observed when flood-pinging on native to a multicast address (e.g. `ping6 100000 ff02::1 0 0`) if there is more than one receiver (e.g. another native node connected over a bridge with an IPv6 address configured -> Linux and native replies to the ping). It remains unclear if this problem exist only on native (or only with the Ethernet netdev2 driver) and if it happens only for multicast pings. I was not able to provoke a full packet buffer, neither on native nor on real hardware using the unmodified `gnrc_networking` example.

cgundogan, 2015-10-05:
>I was not able to provoke a full packet buffer, neither on native nor on real hardware using the unmodified gnrc_networking example.

Did you test this also with (only) 2 neighbors? I just did a quick test with 4 tap devices and multicast ping. The pinging node crashes quite often (not always).  After several attempts, even a node that was pinged crashed.. but I have no idea why right now. Need to investigate, but I guess it has something to do with all those "possibly lost interrupt" messages

kaspar030, 2015-10-05:
> "possibly lost interrupt" messages

In case you continue digging, they occur if the interrupt handler tries 
to send a message indicating there's an interrupt to handle, but 
msg_try_send fails (e.g., the target is not waiting).

authmillenon, 2015-10-05:
Is it crashing or getting full but stays unreleased? These are two very different problem scopes.

authmillenon, 2015-10-05:
Do we really want to milestone this for this release? I don't see it as a blocker for the release, since it is not a very common use case right now (maybe when we have TCP it becomes one)

cgundogan, 2015-10-05:
Crashing

OlegHahm, 2015-10-05:
> Did you test this also with (only) 2 neighbors? 

I tested with 2 to 4 neighbors.

> I just did a quick test with 4 tap devices and multicast ping. The pinging node crashes quite often (not always). After several attempts, even a node that was pinged crashed.. but I have no idea why right now.

Are you sure this is not rather related to https://github.com/RIOT-OS/RIOT/issues/3341 or https://github.com/RIOT-OS/RIOT/issues/3824

> Need to investigate, but I guess it has something to do with all those "possibly lost interrupt" messages

Possibly lost interrupt occurs when the network driver's thread's queue is full.

OlegHahm, 2015-10-05:
> Do we really want to milestone this for this release?

Yes, we do.

> I don't see it as a blocker for the release, since it is not a very common use case right now (maybe when we have TCP it becomes one) 

I don't see it as a blocker, neither. Still a known issue of this release.

authmillenon, 2015-10-05:
Ah, I thought it is a must fix for the release ;-)

authmillenon, 2015-10-05:
With 5 TAPs and the pinging node in GDB (all others just started as binary) I hit https://github.com/RIOT-OS/RIOT/blob/master/cpu/native/native_cpu.c#L144 on the pinging node with `ping6 100000 ff02::1 0 0` at about request seq=38000. The pinged nodes only show `gnrc_netdev2: possibly lost interrupt.` a few (1-4) times

authmillenon, 2015-10-05:
And I just hit `SIGTRAP` again.

authmillenon, 2015-10-05:
https://github.com/RIOT-OS/RIOT/issues/4048#issuecomment-145506696 might point to a scheduler problem in native…

authmillenon, 2015-10-05:
(Doesn't seem to be an issue on `iotlab-m3`, but might also be, that the hardware just isn't fast enough to run into the race condition ^^)

```
1444048548.262235;m3-7;--- ff02::1 ping statistics ---
1444048548.264219;m3-7;10000 packets transmitted, 9945 received, 1% packet loss, time 145.06728901 s
1444048548.265292;m3-7;rtt min/avg/max = 0.387/6.872/49.056 ms
pktbuf
1444048584.182036;m3-11;> pktbuf
1444048584.182864;m3-13;> pktbuf
1444048584.183027;m3-12;> pktbuf
1444048584.183158;m3-7;> pktbuf
1444048584.183269;m3-9;> pktbuf
1444048584.183799;m3-11;packet buffer: first byte: 0x20001c78, last byte: 0x20003478 (size: 6144)
1444048584.184730;m3-11;~ unused: 0x20001c78 (next: 0, size: 6144) ~
1444048584.185397;m3-13;packet buffer: first byte: 0x20001c78, last byte: 0x20003478 (size: 6144)
1444048584.185502;m3-13;~ unused: 0x20001c78 (next: 0, size: 6144) ~
1444048584.185669;m3-12;packet buffer: first byte: 0x20001c78, last byte: 0x20003478 (size: 6144)
1444048584.185766;m3-12;~ unused: 0x20001c78 (next: 0, size: 6144) ~
1444048584.185898;m3-9;packet buffer: first byte: 0x20001c78, last byte: 0x20003478 (size: 6144)
1444048584.186051;m3-9;~ unused: 0x20001c78 (next: 0, size: 6144) ~
1444048584.186183;m3-7;packet buffer: first byte: 0x20001c78, last byte: 0x20003478 (size: 6144)
1444048584.186274;m3-7;~ unused: 0x20001c78 (next: 0, size: 6144) ~
```

cgundogan, 2016-01-05:
@authmillenon could you please confirm whether this problem still persists?

authmillenon, 2016-01-05:
Japp, SIGTRAP on sender, possibly lost interrupt on receiver.

OlegHahm, 2016-01-05:
Isn't this a different problem?

authmillenon, 2016-01-05:
Well the packet buffer is still full of stuff, so I would guess no.

authmillenon, 2016-04-20:
Let's fix this for the next release ;-)

kYc0o, 2016-07-25:
@miri64  It was?

miri64, 2016-07-25:
I honestly don't know. So let's move this again.

kaspar030, 2018-04-20:
Unconfirmed for more than two years -> closing.

closed 2018-04-20 by kaspar030