RIOT-OS/RIOT #4051: ARM toolchain does not format short unsigned.

authmillenon, 2015-10-05:
When I use the `arm-none-eabi` tool chain from [ARM embedded](http://ppa.launchpad.net/terry.guo/gcc-arm-embedded/ubuntu/) `PRIu8` doesn't get printed properly. E.g.:

```
8 bytes from fe80::3432:4833:46d4:8e2a: id=83 seq=9994 hop limit=hu time = 64.003 ms
```

(`hop limit` should be a number).  I'm aware that this is not a RIOT issue, but a toolchain issue, but I was interested if anybody else has this problem.

OlegHahm, 2015-10-05:
With `arm-none-eabi-gcc (Arch Repository) 5.2.0` everything works normal.

authmillenon, 2015-10-05:
Arghs sorry, meant to add my version of the toolchain too: `arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]`

OlegHahm, 2015-10-05:
`arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 224288]` is working, too, so it seems to be broken only for a small range of revisions of this toolchain.

cgundogan, 2015-10-05:
@authmillenon is this maybe related to what you mentioned few days earlier regarding the printing of shorts? Could you maybe try another version of the toolchain and see if this bug is related to your toolchain or maybe anything else in your setup?

authmillenon, 2015-10-05:
> @authmillenon is this maybe related to what you mentioned few days earlier regarding the printing of shorts?

Yes this is related. I just opened the issue only now for this very reason ;-). 

> Could you maybe try another version of the toolchain and see if this bug is related to your toolchain or maybe anything else in your setup?

Well @OlegHahm [did already](https://github.com/RIOT-OS/RIOT/issues/4051#issuecomment-145519841), so only my revision seems to be affected.

closed 2015-10-05 by authmillenon