RIOT-OS/RIOT #4072: doc: Update and provide high-level doc for all the important modules

authmillenon, 2015-10-09:
As discussed in the [meeting from 2015-10-07](http://riot.pad.spline.de/21), we want to update and enhance the current state of the documentation for several of our most important modules. This encompasses high-level documentation of their internals and How-Tos for their internals (as a well documented example). The modules in questions are:

* low-level device drivers (`periph`) => @thomaseichinger 
 - `periph/doc.txt` (general documentation)
    + [ ] high-level doc
    + [ ] How-To
 - `periph/adc.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/cpuid.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/dac.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/gpio.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/i2c.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/pwm.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/random.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/rtc.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/rtt.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/spi.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/timer.h`
    + [ ] high-level doc
    + [ ] How-To
 - `periph/uart.h`
    + [ ] high-level doc
    + [ ] How-To
* generic network stack (`gnrc`) => @cgundogan
 - `net/gnrc.h` (general documentation)
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/ipv6.h` etc.
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/netapi.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/netdev.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/netif.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/netreg.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/nettype.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/pktbuf.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/pkt.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/sixlowpan.h` etc.
    + [ ] high-level doc
    + [ ] How-To
 - `net/gnrc/udp.h`
    + [ ] high-level doc
    + [ ] How-To
* connection API (`conn`) => @authmillenon
 - `net/conn.h` (general documentation)
    + [ ] high-level doc
    + [ ] How-To
 - `net/conn/ip.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/conn/tcp.h`
    + [ ] high-level doc
    + [ ] How-To
 - `net/conn/udp.h`
    + [ ] high-level doc
    + [ ] How-To
 - ... things to come
* RIOT's module system => @OlegHahm
 - [ ] high-level doc
 - [ ] How-To
* RIOT's package system => @Kijewski or @LudwigOrtmann 
 - [ ] high-level doc
 - [ ] How-To
* RIOT's auto-initialization => @OlegHahm 
 - [ ] high-level doc
 - [ ] How-To
* `xtimer` => @kaspar030 
 - [ ] high-level doc
 - [ ] How-To
* generic sensor interface `saul` => @haukepetersen
 - [ ] high-level doc
 - [ ] How-To

The assignments to the modules are people that to my opinion would be best suited to write the documentation. If you don't agree, either change it or let's discuss it. An assignment could also be interpreted that you can sub-delegate the tasks to other people.

emmanuelsearch, 2015-10-09:
Looks good, but I think there is something missing which is key: a point of entry for the high-level documentation, which brings all these elements in the same picture (with coarser granularity if needed). I'm talking about some text and/or graphical elements that would play the role of Section 4 and Figure 1 in http://arxiv.org/pdf/1502.01968v1.pdf for instance.

authmillenon, 2015-10-09:
You mean an update to http://riot-os.org/images/RIOT_network_architecture_dark_updated.png?

cgundogan, 2015-10-09:
What kind of medium should we use for `high-level doc`? While `How-To`s are most probably wiki pages, are `high-level doc`s a combination of doxygen entries, pictures and wiki pages?

authmillenon, 2015-10-09:
In the meeting we pretty much decided, that the end result should be, that the high-level docs are in the doxygen documentation (you can add [images to doxygen](https://www.stack.nl/~dimitri/doxygen/manual/commands.html#cmdimage)) and the how-tos are well-documented example applications in the `examples/`. How we come to that point was a little bit unclear, but it is imaginable, that we start either as a wiki page that than will mature to a doxygen document/example application. The problem with just a wiki page is, that we had the feeling that not so many people use the Wiki as the first point to solve a problem and that it is (by nature) not very structured. That reminds me: I wanted to add an issue for that.

emmanuelsearch, 2015-10-09:
In the meeting (see minutes) we agreed to start with wiki pages, and then in a second phase think about migrating to doxygen.

authmillenon, 2015-10-09:
see #4074 for Wiki.

authmillenon, 2015-10-09:
@emmanuelsearch we did not really agree to that. @kaspar030 was strongly opposed to this.

emmanuelsearch, 2015-10-09:
We can structure the wiki pages the way we want, "by nature" ;)
See for instance the home page which proposes a nice natural point of entry
https://github.com/RIOT-OS/RIOT/wiki
I think the high level documentation we are talking about should ramify from there.

emmanuelsearch, 2015-10-09:
oops. I did not notice that @kaspar030 was opposing wiki to bootstrap the first phase, but maybe I missed something. Anyways, I am convinced that for the point of entry (the big picture bringing "everything together") we should start with a wiki entry, and link it visible in the wiki home https://github.com/RIOT-OS/RIOT/wiki

emmanuelsearch, 2015-10-09:
rest of the documentation could live elsewhere indeed. No strong opinion about this.

authmillenon, 2015-10-09:
I think we should also have a clear understanding (and "rules") about what goes to the wiki, what goes to the doxygen and what goes to README files, so that we (as in people who work with RIOT *and* people who are new to RIOT) don't have to search around for a specific bit of information. Here's what I am thinking about:

* Basic introductory tutorials (how to build an application, how to start a module etc.) to the Wiki (and maybe to the root-README?)
* How to use an example: to the examples README
* Internals and API doc of a module: the module's doxygen documentation
* Board and CPU documentation: Wiki (and doxygen documentation for API documentation)

authmillenon, 2015-10-09:
Packages are currently documented ad different places afaik: some in the Wiki, some in their respective READMEs. Also: some CPUs, as well as some boards, have READMEs in their directory structure, and some don't. What's up with those?

OlegHahm, 2015-10-09:
I think the more goes into Doxygen, the better. How-tos, tutorials, and link collections are great for the Wiki, examples should have a Readme, the rest should go into Doxygen.

authmillenon, 2015-10-09:
> I think the more goes into Doxygen, the better. How-tos, tutorials, and link collections are great for the Wiki, examples should have a Readme, the rest should go into Doxygen.

Then where do you draw the line between how-to, tutorial and example?

OlegHahm, 2015-10-09:
An example has code in the repository, how-tos and tutorials do not.

authmillenon, 2015-10-09:
Didn't we establish in the meeting, that every how-to should be - or at least be accompanied by - an example?

OlegHahm, 2015-10-09:
I don't remember that, but if we did, so what? An example application in the RIOT repository has a README, telling the user what the example does and how to use it. Additional how-tos go to the Wiki.

authmillenon, 2015-10-12:
@emmanuelsearch 

> Looks good, but I think there is something missing which is key: a point of entry for the high-level documentation, which brings all these elements in the same picture (with coarser granularity if [...]

@cgundogan and I discussed this over the last few hours, and we basically came to the conclusion, that text-wise an updated version of the text in [the doxygen](http://doc.riot-os.org/index.html#overview)/[the RIOT introduction in the wiki](https://github.com/RIOT-OS/RIOT/wiki/Introduction) (who are basically two versions of the same text) would suffice. For an introductory graphic we came up that would in the end be pretty similar to http://riot-os.org/images/RIOT_network_architecture_dark_preview.png, just more to the point and focusing on the parts that really need more explanation. This is a first preliminary version of that: 

[image]

emmanuelsearch, 2015-10-12:
that's indeed the sort of thing I was talking about. Looks good to me. Two comments concerning the pic:
- I would probably add 6lowpan to the gnrc zoom so that it does give the "wrong" impression about what this stack is about
- I suppose that we are going to rename netdev2 (to become netdev) once "netdev1" is phased out, right?

authmillenon, 2015-10-12:
> * I would probably add 6lowpan to the gnrc zoom so that it does give the "wrong" impression about what this stack is about

Yes of course, after we were pretty detailed on the left 2/3 of the stack we lost focus and sketched out the rest. The one on the right will in the end be pretty similar to one of the GNRC graphics floating around.

> I suppose that we are going to rename netdev2 (to become netdev) once "netdev1" is phased out, right?

I hope to strip the picture down to a more or less editable SVG we can just include into the doxygen's HTML.

authmillenon, 2015-10-13:
Hier a first step example of how it might look like. I'm about to clean up the SVG to get it in a more readable format (that I will then include into the doxygen).

[image]

gebart, 2015-10-13:
@authmillenon nice! :+1:

emmanuelsearch, 2015-10-13:
looks good

authmillenon, 2015-10-13:
See #4090 for the addition to `doc/`

authmillenon, 2015-10-13:
For the text stuff I started a pad: https://padlite.spline.inf.fu-berlin.de/p/mjtL88j5np

authmillenon, 2015-10-14:
See #4091 how to integrate this.

PeterKietzmann, 2015-12-15:
Hi all! I wanted to support in enhancing the drivers section (adc, pwm, ...) but I got stuck as I don't really know which informations are missing for "externals". So if you already know what **you**  are missing, give me some quick keywords ;-)

authmillenon, 2015-12-15:
This issue can be viewed as more or less deprecated. See #4309 for better overview.

closed 2015-12-15 by authmillenon

PeterKietzmann, 2015-12-15:
good to know...

authmillenon, 2015-12-15:
#4309 was the one linked by @OlegHahm in the maintainer mail, so I'm not sure how you got here ;-)

OlegHahm, 2016-02-03:
Did the above graphic make it into the docs somewhere? I wasn't able to find it.

authmillenon, 2016-02-03:
It has been splited: http://doc.riot-os.org/index.html#structure and http://doc.riot-os.org/group__net__gnrc.html#details

authmillenon, 2016-02-03:
Since the middle part isn't fact in the code base yet we decided against including it.

OlegHahm, 2016-02-03:
Ok, thx for explanation. Btw. isn't the MAC layer missing in the GNRC graphic?

authmillenon, 2016-02-04:
The MAC layer can be handled by `gnrc_netdev2` ;)

authmillenon, 2016-02-04:
(or better said: an extension of it. The current `gnrc_netdev2` is the MAC layer for `netdev2`.

OlegHahm, 2016-02-04:
gnrc_netdev2 doesn't require nomac? But TSCH for example, would still go on top of gnrc_netdev2 IIRC.

authmillenon, 2016-02-04:
No, `gnrc_netdev2` (currently) *IS* `gnrc_nomac` for `netdev2`. Putting a MAC layer on top of it would defy the reason why we made `netdev2` in the first place (no delay due to IPC calls). It should rather go below or instead of `gnrc_netdev2`. I could imagine you can easily overload its `send` and `receive` routines to use for a MAC layer.

OlegHahm, 2016-02-04:
I think I still haven't understand how the lower layers are supposed to work in GNRC.

authmillenon, 2016-02-04:
It changed a little with `gnrc_netdev2`, but just have a look at that module (it's rather slim), then I think you will understand :-).