RIOT-OS/RIOT #4074: doc: clean up the Wiki

authmillenon, 2015-10-09:
According to a quick check-out + `ls *.md | wc -l` we have about 95 wiki pages in our wiki, and it is (at least from my experience) getting harder and harder to actually find stuff that you need in it. Let's discuss how we want to structure the main page and the editing process to make things better there.

OlegHahm, 2015-10-09:
Shouldn't be too difficult. Whenever one add a new Wiki page, link it somewhere. In most cases on the start page. That's what I do (plus from time to time restructuring the hierarchy on the start page). Can you elaborate on what you think is hard to find?

authmillenon, 2015-10-09:
@emmanuelsearch in https://github.com/RIOT-OS/RIOT/issues/4072#issuecomment-146874558
> We can structure the wiki pages the way we want, "by nature" ;)
See for instance the home page which proposes a nice natural point of entry
https://github.com/RIOT-OS/RIOT/wiki
I think the high level documentation we are talking about should ramify from there.

I don't think the current home page of the wiki is very helpful. You basically have to read the whole thing to know where to find stuff. That is also the case if you want to add stuff to it. Example: if I write a how-to, I wouldn't know if I should put it under "Starting the RIOT" or "Developer guide". Also, there are a lot pages, that aren't even linked on the home page. Either, because the author forgot about it, or because it did not make sense for them (or they did not know where to put it maybe).

OlegHahm, 2015-10-09:
> you basically have to read the whole thing to know where to find stuff.

If by "the whole thing" you mean the home page, I think that isn't too much an effort. If you're looking for documentation, you have to read. That's the nature of documentation. ;)

>  Example: if I write a how-to, I wouldn't know if I should put it under "Starting the RIOT" or "Developer guide". 

Just do whatever you think fits best. That was the idea of the Wiki. Let it grow naturely. If someone else thinks, it fits better into another category it can be moved. If one observe flapping, discussion can be triggered.

> Also, there are a lot pages, that aren't even linked on the home page. Either, because the author forgot about it, or because it did not make sense for them (or they did not know where to put it maybe).

If the author forgets to link, there is nothing one can do about it. If isn't linked before one doesn't know where, that seems like a pretty poor reason to me. Either I just put in the best spot I can find, or I add a category (may require restructuring).

emmanuelsearch, 2015-10-09:
OK, so there might be better ways to structure the wiki entries, agreed. 
And we might come up a policy on where to add new entries (hopefully self-explanatory if the structure is good enough).
But that is an intrinsic problem, that has nothing to do with the "Wiki or no Wiki" question for #4072

authmillenon, 2015-10-09:
> Can you elaborate on what you think is hard to find?

I think the best example for that is the installation process for OpenOCD. Since it is not linked on the home page, I always search at the board's documentation first (because they are directly linked in the sidebar it seems to me the most natural place to look for them). Now I go to the page for the IoT Lab M3 and find a link to the OpenOCD project page, but not to the Wiki page (while for the samr21-xpro it is for example). So here for example we should see, if we can somehow unify the pages for the different boards.

Also: from a user perspective the board pages make no sense to me: The first thing popping up are components, board layouts and pin mappings, which from a hardware developers perspective are absolutely crucial, I know. But these are a small percentage of the community, I would say, and the things most people need first, I think are:
* What value do I need to set the `BOARD` variable to to build for this board
* What toolchain do I need (with link about how to install it)
* What flasher do I need (with link about how to install it)

Only after that all the lower level stuff should come, starting with components (and which respective drivers to use), then pin mappings (for GPIO usage), ideally not just linked to some Google Spreadsheet or header file, and then the remaining hardware stuff.

authmillenon, 2015-10-09:
Also also: which board uses which CPU is also something I start searching in the wiki but end up looking up in the Makefiles after all, since for most boards I tend to not find the information (either because it is not there or because its just not at a place obvious to me)

OlegHahm, 2015-10-09:
Well, to me the current structure in both cases makes sense (I don't want to install OpenOCD per se, but only if need for a particular board and I think people want to actually use the board, e.g. connect something to it, so the PIN mappings do matter), but I also don't object re-arranging whatever you feel more comfortable with. Just go ahead.

authmillenon, 2015-10-09:
I'm not asking to arrange the wiki to suit my or your needs, but to find something that everyone is happy with. So just going to the wiki and overhaul it in a way that suits my needs would be the wrong way to do it ;-)

OlegHahm, 2015-10-09:
I disagree.

authmillenon, 2015-10-09:
> so the PIN mappings do matter

I never said they don't matter, I said they don't matter as much for a person who just acquired the board and wants to flash their first RIOT application to this board.

authmillenon, 2015-10-09:
> I disagree.

With what?

OlegHahm, 2015-10-09:
That just going to the wiki and overhaul it to fit one's needs is the wrong way to do it.

authmillenon, 2015-10-09:
Wouldn't that spark some kind of wiki war?

OlegHahm, 2015-10-09:
I doubt it.

authmillenon, 2015-10-10:
Okay I just sat down and tried to get an overview what pages there are and tried to put them in distinct categories. These are the ones I came up with:

* Lists about things that are in RIOT
* Tutorials, Best-practices, FAQ and Cheatsheets (aka the helpful stuff ;-))
* Boards and CPUs
* Statistics
* Organisational stuff (GSoC, future planning, minutes)
* Concepts for parts of RIOT

A-Paul, 2015-10-14:
> Can you elaborate on what you think is hard to find?

@OlegHahm I'm sure @authmillenon can. Because -like you- she knows what to look for.
But, if I don't even know what I have to look for, I'm lost. And I think that is a more common preference for a reader of a wiki -or other documentation.

OlegHahm, 2015-10-23:
Well, then I'm probably wrong person to ask here. ;)

closed 2019-09-10 by stale[bot]