RIOT-OS/RIOT #4075: KSZ8851SNL ethernet driver

poppe34, 2015-10-09:
I am almost finished with the KSZ8851 ethernet driver so you can purchase the ethernet xplain pro expansion board to make the samr21 a border router.  Who else has this board to test and evaluate the driver?  I just want someone to evaluate it other than me.

poppe34, 2015-10-09:
also just a quick thing... Do I add the labels or do you guys.  I could not find them when writing this.

OlegHahm, 2015-10-09:
I added the label, because only maintainers are allowed to set them.

OlegHahm, 2015-10-09:
btw. the expansion board looks pretty expansive.

poppe34, 2015-10-09:
Dont disagree with you there.  But it seems like the Samr21 is used alot on here.  Since I already had the board I wanted to use it for my BR.  My next project will to be to add the TI1294 ethernet board.  I wanted to get my feet wet with the ethernet driver first so I can get use to the new netdriver.  That board is super cheap.  I just need to get/make a board that has an 802.15.4 transceiver.

OlegHahm, 2015-10-22:
Hey @poppe34, actually, by chance I got hold of an [Atmel Ethernet1 Xplained Pro](http://www.atmel.com/tools/ETHERNET1_XPRO.aspx). Is this the one featuring the mentioned chip? In this case I would be glad to test.

poppe34, 2015-10-22:
Yes it is. Awesome I squashed a couple bugs and really close to making the pull request

OlegHahm, 2015-10-22:
Cool, I'm very excited about and eager to test this. Will be great to have Ethernet-capable border router.

iseitani, 2015-10-30:
Hello, can you help me flash a r21 board as a border router?
Thank you!

poppe34, 2015-10-30:
I would be happy to help out.  I have made a pull request with the needed source code.  But I have made a few rookie mistakes in GIT and I will be resubmitting the PULL request.  I will make sure that the test readme file has an outline on how to make it a boarder router.  I might even make a pull request updating the boarder router app on how to do it once the pull is merged.  Feel free to email me if you have any other questions.

iseitani, 2015-10-30:
I tried to flash [this](https://github.com/RIOT-OS/RIOT/blob/master/examples/gnrc_border_router/README.md) by changing the USEMODULE value but nothing :(

poppe34, 2015-10-30:
It has not been merged into the main branch as of yet.  If you go to my Fork of RIOT you can go to the Ethernet XPRO branch and see the changes and even test some of them.  I have not fully tested the driver yet and may not work 100%.  I am planning on working on it this afternoon

iseitani, 2015-10-30:
I know it sounds a little bit strange , but can you send me the link of the branch?

msolters, 2015-11-02:
Ahoy!  I'll definitely be looking into this :+1:

iseitani, 2015-11-02:
@poppe34  in ksz8851_netdev.c in line 27 : #include "ksz8851snl.h" .Where is this file located?
Thank you!

haukepetersen, 2015-11-02:
By the way, if anyone is looking for a cheaper way to bring ethernet to your favorite node: have a look at the enc28j60 ethernet modules (they are around $2...). Driver is now available: #4200

msolters, 2015-11-02:
@iseitani it is located inside `drivers/include`, where all driver headers are located.  [Examining the PR](https://github.com/RIOT-OS/RIOT/pull/4144) would have provided you this information, either by looking [commit that contains the files](https://github.com/RIOT-OS/RIOT/commit/ae2222bc6683bb1c3f1ca8c78975f38f04a135ec), or by merging and simply searching your work tree.

iseitani, 2015-11-03:
@msolters  it's my mistake for not looking inside the drivers/include ,but the .h files are not there in all OS. Merge and search my tree??? Today I was able to fork the ksz8851 driver, 2 braches untill yesterday , 3 today  ;)

msolters, 2015-11-03:
@iseitani all I mean is a general solution to the problem "where is file X?" is usually best approached by searching through (1) the remote commit history or (2) a local copy in your own file system!

Even when a branch is destroyed by the owner, the PR will still retain a copy of the files in the commit history here on GitHub until the PR itself is destroyed.

msolters, 2015-11-03:
We do have a clean(ish) PR to discuss this on now -- maybe we can close this, no?

haukepetersen, 2015-11-03:
@msolters: make sense to close this issue. I would suggest for technical discussions regarding the driver to continue there #4212 and to move discussions on the setup of a ethernet-15.4-boarder router to the devel mailing list.

closed 2015-11-03 by haukepetersen