RIOT-OS/RIOT #4087: SAMR21: interrupt not disabled when gnrc_ipv6_router_default enabled

bapclenet, 2015-10-13:
Here is the main and the Makefile to reproduce the error;

main.c

    #include <stdio.h>
    #include "periph/gpio.h"
    
    void callback(void){
        printf("INT callback\n");
    }
    
    int main(void) {
        gpio_init_int(GPIO(PA,28), GPIO_PULLUP, GPIO_FALLING, (gpio_cb_t)callback, NULL);
        gpio_irq_disable(GPIO(PA,28));
        return 0;
    }

Makefile

    # name of your application
    APPLICATION = int_example
    
    # If no BOARD is found in the environment, use this default:
    BOARD ?= native

    # This has to be the absolute path to the RIOT base directory:
    RIOTBASE ?= $(CURDIR)/../..

    BOARD_INSUFFICIENT_RAM := airfy-beacon chronos msb-430 msb-430h nrf51dongle \
                          nucleo-f334 pca10000 pca10005 redbee-econotag \
                          stm32f0discovery telosb wsn430-v1_3b wsn430-v1_4 \
                          yunjia-nrf51822 z1
    # Include packages that pull up and auto-init the link layer.
    # NOTE: 6LoWPAN will be included if IEEE802.15.4 devices are present
    USEMODULE += gnrc_netif_default
    USEMODULE += auto_init_gnrc_netif
    # Specify the mandatory networking modules for IPv6 and UDP
    USEMODULE += gnrc_ipv6_router_default

    CFLAGS += -DDEVELHELP

    # Change this to 0 show compiler invocation lines by default:
    QUIET ?= 1

    include $(RIOTBASE)/Makefile.include

Flash on the SAMR21. Then after the interuption gets disabled, press SW0 and wait at least 10 seconds, you should see that the interruption is raised! And this is like it waits every 10 secs after having disabled to check if an interrupt has raised before calling the callback.
Example

Time | Int Enable  | SWO pressed | Int raised
-----|------------ | -------------| -------------
0|1|0|0
1|0|0|0
2|0|0|0
3|0|0|0
4|0|1|0
5|0|0|0
6|0|0|0
7|0|0|0
8|0|0|0
9|0|0|0
10|0|0|0
11|0|0|1
12|0|0|0

If you remove 

    USEMODULE += gnrc_ipv6_router_default

This weird behaviour won't happen.
Can you see the same behaviour?

I tried with the test example on GPIO by adding a disable_int function and it works as it should so I suspect gnrc_ipv6_router_default to do something that it shouldn't.

daniel-k, 2015-10-15:
First thing to check is if you unintentionally overflow the ISR stack. Using printf inside an interrupt is always really dangerous.

daniel-k, 2015-10-15:
The issue was caused by the INT-line of the `at86rf233` transceiver that will be automagically initialized when you add `gnrc_ipv6_router_default`. This triggered an interrupt from time to time and then caused what will be fixed by #4098.

bapclenet, 2015-10-16:
I know for the printf :-), was just for a quick debug.
You did a great job with this PR! 
Thanks for having worked on it.

Why 
>  This triggered an interrupt from time to time

?
If we make the processor sleep and not the transceiver, it is going to be waken up every ten seconds basically?

closed 2015-10-16 by bapclenet

daniel-k, 2015-10-16:
> If we make the processor sleep and not the transceiver, it is going to be waken up every ten seconds basically?

Where and how do you put the processor to sleep?

If the transceiver is active, depending on it's mode, even invalid incoming frames might trigger an interrupt. I have boards lying around that send broadcasts regularily, I guess that's why I've had the same behaviour.

bapclenet, 2015-10-16:
> Where and how do you put the processor to sleep?

As soon as LPM works :-). Yes I see, if I do not want any wake up from the transceiver I will put in sleep mode.

commit 1f98e069 (2015-10-16): samd21/gpio: check if interrupt is enabled in isr, fixes #4087

The interrupt flag for a previously configured external gpio interrupt
will be set regardless of the actual configuration. So when another source
causes an interrupt, the callback of a disabled gpio interrupt will be serviced
although it was disabled if the interrupt occured in the meantime.
commit c652fd8d (2015-10-16): Merge pull request #4098 from daniel-k/pr/samd21_gpio_irq

samd21/gpio: check if interrupt is enabled in isr, fixes #4087