RIOT-OS/RIOT #4095: unittests/tests-sixlowpan_ctx: Missing call to xtimer_init

gebart, 2015-10-15:
Hard fault on K60/mulle because of trying to read an uninitialized timer peripheral.

It doesn't crash in any example application because of auto_init for the xtimer module.

The solution is to initialize the xtimer module before the test, but I don't know the consequences of that since the unittests have disabled auto-init for a reason.

backtrace:

```
#0  0x0001a31c in pit_timer_read (ch=1 '\001', dev=0) at /data/riotbuild/riotcpu/kinetis_common/timer.c:81
#1  timer_read (dev=dev@entry=0) at /data/riotbuild/riotcpu/kinetis_common/timer.c:224
#2  0x00019efe in _xtimer_now () at /data/riotbuild/riotbase/sys/include/xtimer.h:409
#3  0x00019f22 in xtimer_now () at /data/riotbuild/riotbase/sys/include/xtimer.h:468
#4  _xtimer_now64 (long_term=<optimized out>, short_term=<optimized out>) at /data/riotbuild/riotbase/sys/xtimer/xtimer_core.c:71
#5  xtimer_now64 () at /data/riotbuild/riotbase/sys/xtimer/xtimer_core.c:84
#6  0x00019e86 in vtimer_now (out=out@entry=0x1fff0c30 <main_stack+1272>)
    at /data/riotbuild/riotbase/sys/compat/vtimer/vtimer_compat.c:44
#7  0x000187e8 in _current_minute () at /data/riotbuild/riotbase/sys/net/gnrc/network_layer/sixlowpan/ctx/gnrc_sixlowpan_ctx.c:136
#8  gnrc_sixlowpan_ctx_update (id=id@entry=5 '\005', prefix=prefix@entry=0x1fff0c58 <main_stack+1312>, 
    prefix_len=prefix_len@entry=63 '?', ltime=ltime@entry=10098, comp=comp@entry=true)
    at /data/riotbuild/riotbase/sys/net/gnrc/network_layer/sixlowpan/ctx/gnrc_sixlowpan_ctx.c:127
#9  0x00005af0 in test_sixlowpan_ctx_update__success ()
    at /data/riotbuild/riotproject/tests/unittests/tests-sixlowpan_ctx/tests-sixlowpan_ctx.c:55
#10 0x00005e68 in test_sixlowpan_ctx_update__wrong_id1 ()
    at /data/riotbuild/riotproject/tests/unittests/tests-sixlowpan_ctx/tests-sixlowpan_ctx.c:77
#11 0x00016d00 in TestCase_run (self=0x1fff0cc4 <main_stack+1420>, result=0x1fff1e74 <result_>)
    at /data/riotbuild/riotbase/sys/embunit/TestCase.c:58
#12 0x00016b4c in TestCaller_run (self=0x1f0e0 <sixlowpan_ctx_tests>, result=0x1fff1e74 <result_>)
    at /data/riotbuild/riotbase/sys/embunit/TestCaller.c:54
#13 0x0000046a in main () at /data/riotbuild/riotproject/tests/unittests/main.c:30
```

authmillenon, 2015-10-15:
I would just put it into the beginning of the `unittests` main function. `_init()` functions can't be tested by this unittest framework anyways, since it does not reset the node after every test.

authmillenon, 2015-10-15:
The side-effects should be minimal to non-existent for the unittests, if I see this correctly.

PR #4100 (2015-11-06): unittests: initialize xtimer if present
Fixes #4095.
closed 2015-11-06 by gebart