RIOT-OS/RIOT #4097: at86rf2xx: potential unaligned access

daniel-k, 2015-10-15:
The NETAPI interface has lots of memory accesses like this:
```
*((uint16_t *)val) = at86rf2xx_get_txpower(dev);
```

I see that this is convenient, but can easily cause hardfaults because of unaligned access. Do we assume that the user only passes aligned memory if his platform doesn't support unaligned access? Either way this should be documented.

thomaseichinger, 2015-10-15:
You are absolutely right, we don't mention it and we probably should. Did you ever experience this? As far as I can see, unaligned access would happen if (i) `val` points to a smaller type (in the upper case e.g. `int8_t`) or (ii) the memory written to is in a packed structure. Do you see other possibilities? (Both upper cases are not a good idea to use with `get` and `set` anyway though.)

daniel-k, 2015-10-15:
@thomaseichinger 
Found this while I passed a member of packed structure to `get`. It worked for a long time, then I changed the order and it crashed.

Another (but similar) possibility would be if you pass (smaller typed) memory in an array. There's also an `*(uint64_t*)...` write like this, but I suspect this this would be splitted into smaller writes on most platforms.

thomaseichinger, 2015-10-15:
Yep, these are exactly the two cases described above. I think we should mention this in the documentation that these functions don't check for alignment and the MAC protocol implementation is responsible to assure alignment of passed memory to access. Because there is no reasonable way to check for this in `get` and `set`. Do you agree?

daniel-k, 2015-10-16:
> Do you agree?
Yes.

PR #4101 (2015-10-16): doc:netdev: mention get/set don't check for alignment of memory accessed
fixes #4097
closed 2015-10-16 by daniel-k