RIOT-OS/RIOT #4105: error : No source available for "0x1fff37b8

ksdperera, 2015-10-18:
Hi All,

I just follow this tutorial : https://github.com/…/Using-the-Eclipse-IDE-for-C-and-CPP-De…

for stm32f4discovery board.
flashing work fine but when I'm trying to debug I got this error : No source available for "0x1fff37b8".

Plz someone who are familiar about RIOT OS use with Eclipse IDE,
I need a big favor from you!

Thank you

OlegHahm, 2015-10-20:
@BytesGalore, @josephnoir, any ideas?

DipSwitch, 2015-10-20:
Please validate the BOOT0 and BOOT1 configuration, it seems the processor is running from the System memory + OTP section.

BytesGalore, 2015-10-20:
@DipSwitch thx for the hint, I will try to reproduce the misbehaviour as you described tomorrow (I don't have the (no) board around now)

ksdperera, 2015-10-22:
@DipSwitch thanx for the solution, It was debug without error after done the following BOOT0 and BOOT1 configuration. 
BOOT1 = x BOOT0 = 1  Main flash memory selected as boot space.
[image]

closed 2015-10-22 by ksdperera