RIOT-OS/RIOT #4113: Bug in NDP code

daniel-k, 2015-10-20:
I'm not too familiar with this part of RIOT but was getting hardfaults when setting up `example/gnrc_networking` @ master on a `samr21-xpro` board. When I try to add an address, i.e. `ifconfig 7 add affe::7`, it hardfaults.

It hardfaults inside memcpy, called from [here](https://github.com/RIOT-OS/RIOT/blob/4e90fdf5059afcfcea08e6e47a23e6f28ae60ecb/sys/net/gnrc/network_layer/ndp/internal/gnrc_ndp_internal.c#L626) because `sl2a_len` is `0xfffffffe`. That's caused from the [code above](https://github.com/RIOT-OS/RIOT/blob/4e90fdf5059afcfcea08e6e47a23e6f28ae60ecb/sys/net/gnrc/network_layer/ndp/internal/gnrc_ndp_internal.c#L619-L626) where `sl2a_opt->len` evaluates to `2`, so `(sl2a_opt->len / 8)` is `0` and therefore `(sl2a_opt->len / 8) - sizeof(ndp_opt_t)` gets `0xfffffffe` (unsigned -2).

As I don't know what's happening there I didn't attempt to fix it myself. If you need further input please ask.

OlegHahm, 2015-10-20:
Currently, I'm not able to reproduce it:
```
2015-10-20 15:42:03,711 - INFO # main(): This is RIOT! (Version: 2014.12-3836-g4e90f-orchid)
2015-10-20 15:42:03,714 - INFO # RIOT network stack example application
2015-10-20 15:42:03,717 - INFO # All up, running the shell now
ifconfig 7 add affe::7
2015-10-20 15:42:05,258 - INFO # > ifconfig 7 add affe::7
2015-10-20 15:42:05,262 - INFO # success: added affe::7/64 to interface 7
ifconfig
2015-10-20 15:42:07,734 - INFO # > ifconfig
2015-10-20 15:42:07,743 - INFO # Iface  7   HWaddr: 20:3a  Channel: 26  NID: 0x23  TX-Power: 0dBm  State: IDLE CSMA Retries: hu 
2015-10-20 15:42:07,747 - INFO #            Long HWaddr: 5a:41:20:49:96:19:20:3a 
2015-10-20 15:42:07,752 - INFO #            AUTOACK  CSMA  MTU:1280  6LO  IPHC  
2015-10-20 15:42:07,755 - INFO #            Source address length: 8
2015-10-20 15:42:07,758 - INFO #            Link type: wireless
2015-10-20 15:42:07,763 - INFO #            inet6 addr: ff02::1/hu  scope: local [multicast]
2015-10-20 15:42:07,769 - INFO #            inet6 addr: fe80::5841:2049:9619:203a/hu  scope: local
2015-10-20 15:42:07,775 - INFO #            inet6 addr: ff02::1:ff19:203a/hu  scope: local [multicast]
2015-10-20 15:42:07,779 - INFO #            inet6 addr: affe::7/hu  scope: global
2015-10-20 15:42:07,785 - INFO #            inet6 addr: ff02::1:ff00:7/hu  scope: local [multicast]
```

daniel-k, 2015-10-20:
This happens only when a boarder router (same board) is present.

OlegHahm, 2015-10-20:
Looking at the code I assume that it will happen always if some neighbor sends a NS with SL2AO being set.

authmillenon, 2015-10-20:
Seems like this line is wrong. https://tools.ietf.org/html/rfc4861#section-4.6.1 states that the length field is in units of 8 bytes so `sl2a_len` (which is in byte) should be

```C
sl2a_len = (sl2a_opt->len * 8) - sizeof(ndp_opt_t);
```

Can anybody confirm that?

OlegHahm, 2015-10-20:
I don't understand why the condition `(sl2a_len == 0)` is true in the first place. Apparently, there is no netif header present.

OlegHahm, 2015-10-20:
and it seems to be some Heisenbug: as soon as I enable some debugging, it disappears.

OlegHahm, 2015-10-20:
For some reason https://github.com/RIOT-OS/RIOT/blob/4e90fdf5059afcfcea08e6e47a23e6f28ae60ecb/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L706 evaluates to `false` in some cases.

OlegHahm, 2015-10-20:
> Seems like this line is wrong. https://tools.ietf.org/html/rfc4861#section-4.6.1 states that the length field is in units of 8 bytes so sl2a_len (which is in byte) should be
```
sl2a_len = (sl2a_opt->len * 8) - sizeof(ndp_opt_t);
```
> Can anybody confirm that?

I think you're right.

authmillenon, 2015-10-20:
> For some reason https://github.com/RIOT-OS/RIOT/blob/4e90fdf5059afcfcea08e6e47a23e6f28ae60ecb/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c#L706 evaluates to false in some cases.

Maybe 6LoWPAN does something wrong :/

OlegHahm, 2015-10-20:
Maybe the observation stated above was wrong. But it seems that NDP receives a pointer to another snip in the packet buffer as the `_receive()` function in IPv6 in the error case.

OlegHahm, 2015-10-20:
Seems like a riddle: `gnrc_icmpv6_demux()` is called twice, after only one IPv6 `_receive()` call...

OlegHahm, 2015-10-20:
Ah, sorry, forgot the two last comments.

OlegHahm, 2015-10-27:
Sorry, I haven't had much time to debug this further lately and probably won't have much time during the next weeks.

I think the problem is two-fold: 
1. the calculation of the size is wrong as @authmillenon pointed out. Fixing this should fix the hardfault.
2. this function should not be called (at least not in 6LoWPAN context) without a pointer to the netif header. I have no idea why this happens.

OlegHahm, 2015-10-27:
https://github.com/RIOT-OS/RIOT/pull/4179 should address 1. @daniel-k, can you test, please?

daniel-k, 2015-10-27:
Will do thanks, thanks!

authmillenon, 2016-04-20:
The border router was overhauled quite significantly. Is this still an issue.

miri64, 2016-10-18:
Ping?

OlegHahm, 2017-01-15:
Timeout.

closed 2017-01-15 by OlegHahm

daniel-k, 2017-01-15:
Oops, looks like I was talking about a very distant "tomorrow" 😄 . Sorry, totally lost track of this issue. Since this is already 1.5y in the past and I don't have any boards to test let's hope it's fixed now. Thanks anyway @OlegHahm and @miri64.