RIOT-OS/RIOT #4125: cortex-m: build error in hard fault handler with -O0

OlegHahm, 2015-10-21:
When building with `-O0` (`CFLAGS_OPT=-O0`) compiling of the cortexm_common code fails:
```
/home/oleg/git/RIOT/cpu/cortexm_common/vectors_cortexm.c: In function 'hard_fault_default':
/home/oleg/git/RIOT/cpu/cortexm_common/vectors_cortexm.c:172:1: error: r7 cannot be used in asm here
 }
 ^
```

OlegHahm, 2015-10-21:
see https://tls.mbed.org/kb/development/arm-thumb-error-r7-cannot-be-used-in-asm-here

gebart, 2015-10-21:
is it possible to `#ifdef`?

Kijewski, 2015-10-21:
I had the same problem for x86. See e.g. [x86_ucontext.c#L35](https://github.com/RIOT-OS/RIOT/blob/7b10e59ef3ef37cde032ddcc4c25b74f781d7428/cpu/x86/x86_ucontext.c#L35) how override the optimization level locally.

daniel-k, 2015-10-21:
What about:
```
mov r4, r8
mov r5, r9
mov r6, r10
push {r4-r6}
push r7          
```

We don't care too much about performance here anyway.

gebart, 2015-10-21:
is the error message caused by the actual asm statements or is it caused by listing `r7` in the clobber list?

gebart, 2015-10-21:
looking at the code it seems like r0 should be listed in the clobber list as well, it won't work if one of the inputs is placed in r0

gebart, 2015-10-21:
I made a quick fix, please try #4126

closed 2015-10-28 by gebart

OlegHahm, 2015-10-28:
:bow: